pop-os/system76-power #237: Test Coverage

danieleades, 2021-04-16:
Is there any interest in tracking test coverage for this crate?

i'm aware that the majority of testing done for this application is necessarily manual integration testing, so it can therefore be a non-goal of this crate to achieve a high percentage of test coverage.

That said, there are a lot of functions that could benefit from unit tests, and monitoring test coverage provides a useful mechanism to ensure that *new* code which is added has a reasonable compliment of unit tests to show that it does what it says it does.
Could also be used to catch API regressions/changes, for example by testing that command line argument strings can be parsed.

I'm proposing this be added as a non-failing github action. probably using `tarpaulin` + `codecov` to add coverage comments on pull requests. The coverage can be totally ignored of course, but it's a nice, semi-passive way to encourage the amount of test coverage to ratchet up over time.

thoughts?

crawfxrd, 2021-04-16:
Note: Codecov announced yesterday that the Bash Uploader script had been compromised [\[1\]](https://about.codecov.io/security-update/). The GitHub Action should not be used until codecov/codecov-action#281 is resolved.

danieleades, 2021-04-16:
> Note: Codecov announced yesterday that the Bash Uploader script had been compromised [[1]](https://about.codecov.io/security-update/). The GitHub Action should not be used until [codecov/codecov-action#281](https://github.com/codecov/codecov-action/issues/281) is resolved.

good shout. i got quite a scary looking email about this yesterday. Should expect to see a fix within days (looks like a small change).

danieleades, 2021-04-16:
> Note: Codecov announced yesterday that the Bash Uploader script had been compromised [[1]](https://about.codecov.io/security-update/). The GitHub Action should not be used until [codecov/codecov-action#281](https://github.com/codecov/codecov-action/issues/281) is resolved.

This has now been merged

PR #238 (2021-09-21): add coverage workflow
closes #237

the coverage of this repo is low, but that doesn't mean coverage testing isn't useful. it provides a mechanism for ensuring that coverage at least improves over time
closed 2021-09-21 by jackpot51