pop-os/system76-power #240: System76 Power Management extension fails to load

pghpete, 2021-04-21:
At some point my "System76 Power Management" gnome extension stopped loading and now I do not have the ability to change between my internal GPU and my Nvidia GPU on my Gazelle. gaze15, I have reinstalled both system76-power and gnome-shell-extension-system76-power to no avail.

Opening the gnome extensions front end I see the following...

```
Description: System76 Power Management
Error: GioDBusError:
Gio.DBusError: GDBus.Error:org.freedesktop.DBus.Error.AccessDenied: Rejected send message, 1 matched rules; type="method_call", sender=":1.83" (uid=1010 pid=2335 comm="/usr/bin/gnome-shell " label="unconfined") interface="com.system76.PowerDaemon" member="GetProfile" error name="(unset)" requested_reply="0" destination=":1.3" (uid=0 pid=1176 comm="/usr/bin/system76-power daemon " label="unconfined")
```

**Distribution (run `cat /etc/os-release`):**
 ```
 cat /etc/os-release
NAME="Pop!_OS"
VERSION="20.10"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 20.10"
VERSION_ID="20.10"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=groovy
UBUNTU_CODENAME=groovy
LOGO=distributor-logo-pop-os

```
 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
```
[$ sudo apt list --installed system76-power
Listing... Done
system76-power/groovy,now 1.1.16~1618506466~20.10~214e1c8 amd64 [installed,automatic]
]$ sudo apt list --installed gnome-shell-extension-system76-power
Listing... Done
gnome-shell-extension-system76-power/groovy,groovy,now 2.0.1~1614194984~20.10~bb901df all [installed,automatic]

$ apt policy system76-power
system76-power:
  Installed: 1.1.16~1618506466~20.10~214e1c8
  Candidate: 1.1.16~1618506466~20.10~214e1c8
  Version table:
 *** 1.1.16~1618506466~20.10~214e1c8 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu groovy/main amd64 Packages
        100 /var/lib/dpkg/status
$ apt policy gnome-shell-extension-system76-power
gnome-shell-extension-system76-power:
  Installed: 2.0.1~1614194984~20.10~bb901df
  Candidate: 2.0.1~1614194984~20.10~bb901df
  Version table:
 *** 2.0.1~1614194984~20.10~bb901df 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu groovy/main amd64 Packages
       1001 http://ppa.launchpad.net/system76/pop/ubuntu groovy/main i386 Packages
        100 /var/lib/dpkg/status

```

**Issue/Bug Description:**

System76-power and gnome-shell-extension-system76-power refuse to load with the following Gio.DBusError 

```
Description: System76 Power Management
Error: GioDBusError:
Gio.DBusError: GDBus.Error:org.freedesktop.DBus.Error.AccessDenied: Rejected send message, 1 matched rules; type="method_call", sender=":1.83" (uid=1010 pid=2335 comm="/usr/bin/gnome-shell " label="unconfined") interface="com.system76.PowerDaemon" member="GetProfile" error name="(unset)" requested_reply="0" destination=":1.3" (uid=0 pid=1176 comm="/usr/bin/system76-power daemon " label="unconfined")
```

**Steps to reproduce (if you know):**
None, necessary as it isn't working as-is.

**Expected behavior:**
The extension loads properly so I have the ability to change between intel/nvidia or hybrid mode as I was able to when I bought my Gazelle15.

**Other Notes:**
I would be grateful for any assistance and would be happy to help with any logs, commands, and debugging necessary to get this running properly once more.
A screenshot of the extensions gui with the error has been attached to this issue.

[image]

bflanagin, 2021-04-22:
@pghpete; I'm unable to replicate the issue on my gaze15 exactly as you see it.

Somethings you might try:
1. Purge and reinstall your nvidia driver (the issue may be tied to it)
```
sudo apt purge nvidia-*
sudo apt install system76-driver-nvidia

``` 
Once done, restart the laptop.

2. Create a test account: Sometimes the users local settings can cause issues with various extensions. Creating a test account (and logging in as them) might determine if the issue is system wide or specific to your account. 

If neither of these work I suggest putting in a support ticket at https://support.system76.com. They answer faster and can work with you more directly than we can here on github.

leviport, 2021-04-22:
See these instructions for reinstalling Nvidia drivers: https://support.system76.com/articles/login-loop-pop/#reinstall-nvidia-driver

That `nvidia-*` wildcard won't work the way you want it to. Purging `~nnvidia` is what you want.

If that doesn't help, you may also want to reinstall `system76-driver` and all of its various dependencies.

pghpete, 2021-04-22:
Thanks for the suggestions folks. I did the following...
sudo apt purge ~nvidia
sudo apt install system76-driver-nvidia

(and it didn't help unfortunately)

I then added an additional user and logged as that user but it has the same issue.

I'll keep digging and see what I can find and update this post with any updates.

pghpete, 2021-04-22:
I have found another, bigger issue I think may be contributing to this...

It appears my system is properly installing new kernels during updates but is NOT booting to the newer ones, rather it is sticking to the old one. (despite many reboots)

```
$ ls -1 /boot | grep -E "^v|^i"
initrd.img
initrd.img-5.11.0-7612-generic
initrd.img-5.3.0-7648-generic
initrd.img-5.8.0-7642-generic
initrd.img.old
vmlinuz
vmlinuz-5.11.0-7612-generic
vmlinuz-5.8.0-7642-generic
vmlinuz.old

$ uname -r
5.8.0-7642-generic
```
I don't know much about systemd-boot vs grub so I'm going to have to dig into that now.

pghpete, 2021-04-22:
Ok so I figured out my kernel issue...

I am using btrfs in a raid1/mirror setup following instructions from Willi here: https://mutschler.eu/linux/install-guides/pop-os-btrfs-raid1/ 

While my system is updating properly it appears that at boot the "backup" or mirror /boot/efi drive/partition was being  seen instead of the primary with all of the updated files.  I solved this temporarily by mounting the mirror (nvme1n1p1) and rsyncing /boot/efi to /boot-efi-mirror

[root@sys76: ~]$ lsblk -f| grep -B 1 FAT32
nvme0n1                                                                                               
├─nvme0n1p1          vfat        FAT32          D01A-F492                               186.3M    63% /boot/efi
--
nvme1n1                                                                                               
├─nvme1n1p1          vfat        FAT32          D01A-F492                               186.3M    63% /boot-efi-mirror
                                       
Once I did this and rebooted the system booted to the 5.11 kernel instead of the 5.8 as it had been.  I will just have to remember to do this procedure when the kernel/initrd are updated and/or rebuilt. Or track down what within systemd-boot is calling on the wrong drive/partition and fix it there.

After this,  I assuming some of my other updated packages and modules may not have built properly on the 5.11 since only the 5.8 was "running".

I then repeated the following...

sudo apt purge ~nvidia
sudo apt install system76-driver-nvidia

rsynced /boot/efi again and rebooted.  My system then prompted for an OS update and an nvidia update. I applied those, repeated my rsync and rebooted yet again.  While this solved my kernel issue it has not solved my systemd-power issue.

I feel like I'm closer to figuring this out but I am not at that point yet.

pghpete, 2021-05-06:
At some point my default user on my system76 laptop was removed from the "adm" group.  Adding the user back to that group using `gpasswd -a <userid> adm` and rebooting did the trick.  I got the idea from another post [https://github.com/pop-os/system76-power/issues/170](https://github.com/pop-os/system76-power/issues/170)

Thanks to all that have helped me along the way.  I will close this and mark it "solved".

closed 2021-05-06 by pghpete