pop-os/system76-power #241: daemon returned an error message: "does not have switchable graphics"

rdgh8, 2021-04-27:
**Distribution (run `cat /etc/os-release`):**
 
NAME="Pop!_OS"
VERSION="20.10"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 20.10"
VERSION_ID="20.10"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=groovy
UBUNTU_CODENAME=groovy
LOGO=distributor-logo-pop-os
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**

system76-power version 1.1.16

**Issue/Bug Description:**

I cannot switch graphics modes on my laptop. I also don't see the extension loaded in the menu under the wifi and bluetooth settings, even though it is installed and turned on in Gnome Extensions. 

**Steps to reproduce (if you know):**

1. Run `sudo system76-power graphics compute` in the terminal
```
setting graphics to compute
daemon returned an error message: "does not have switchable graphics"
```

**Expected behavior:**

Graphics mode switches to compute. 

**Other Notes:**

If it changes anything, the laptop does not have a battery installed and runs plugged in on wall power 100% of the time. The GPU is a GTX 1060 6 GB with the newest 460.67 drivers installed. 

https://user-images.githubusercontent.com/66289396/116318926-99557100-a783-11eb-94ce-cb774ee9d72b.mp4

crawfxrd, 2021-04-27:
What's the output of `lspci | grep VGA`?

If there's not 2 devices, then it doesn't have switchable graphics.

rdgh8, 2021-04-27:
`01:00.0 VGA compatible controller: NVIDIA Corporation GP106M [GeForce GTX 1060 Mobile] (rev a1)`

However, I have an i7 8750h, this CPU has a UHD 630?

crawfxrd, 2021-04-27:
So what is likely the case is that the eDP and all display ports are wired to the NVIDIA GPU, leaving the integrated graphics disabled.

closed 2021-04-30 by crawfxrd

alierkan, 2023-01-07:
>lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation CometLake-S GT2 [UHD Graphics 630] (rev 05)
01:00.0 VGA compatible controller: NVIDIA Corporation GA104 [GeForce RTX 3060] (rev a1)

But I got the same error, and I couldn't switch. Obviously, there is a problem related to UHD 630.