pop-os/system76-power #242: [Feature Request] List target graphics mode

MilesBHuff, 2021-04-30:
Currently, running `system76-power graphics` shows only the current graphics mode.  [This has led to confusion online](https://aur.archlinux.org/packages/system76-power/#comment-796800).

**Suggestion:  Show the target graphics mode as well.**

### Current operation:
```
$ system76-power graphics
hybrid
$ system76-power graphics integrated
$ system76-power graphics
compute
$ systemctl reboot
$ system76-power graphics
integrated
$ system76-power graphics hybrid
$ system76-power graphics
integrated
```

### Desired operation:
```
$ system76-power graphics
hybrid
$ system76-power graphics integrated
$ system76-power graphics
compute (integrated after reboot)
$ systemctl reboot
$ system76-power graphics
integrated
$ system76-power graphics hybrid
$ system76-power graphics
integrated (hybrid after reboot)
```

crawfxrd, 2021-04-30:
I think it just needs to be explicitly documented (README  and `system76-power graphics -h`) that a reboot is required after switching modes for changes to take effect.

MilesBHuff, 2021-04-30:
That would also be a good thing.  I'm surprised it isn't already.

crawfxrd, 2021-04-30:
Would you consider #246 sufficient? It will now inform the user that a reboot is required when they change graphics modes.

MilesBHuff, 2021-05-01:
Yeah, that seems pretty much perfect.  It would be great if a message could also be added to the getter *if it is pending a reboot*, though.

MilesBHuff, 2021-05-01:
Actually, something else just occurred to me:  hybrid -> compute is possible without a reboot.  The warning messages in that PR don't account for that.

closed 2021-05-03 by bflanagin