pop-os/system76-power #243: [Documentation] Mention `prime-run`;  raw variables are unwieldly

MilesBHuff, 2021-04-30:
**Request to mention the existence of `prime-run`, which is a script that comes with `nvidia-prime`.**

Currently, the documentation says this:
> Vulkan applications must be launched with `__NV_PRIME_RENDER_OFFLOAD=1` to render on the dGPU. GLX applications must be launched with `__NV_PRIME_RENDER_OFFLOAD=1` `__GLX_VENDOR_LIBRARY_NAME=nvidia` to render on the dGPU.

While true, this is highly unwieldly for commandline users.  Adding the following line to the above would help usability significantly:
> The `prime-run` script included with `nvidia-prime` sets these variables for you;  meaning you can simply run `prime-run $APP` to start an application with the dGPU.

MilesBHuff, 2021-04-30:
Issue implemented in PR #244.

crawfxrd, 2021-04-30:
`system76-power` conflicts with `nvidia-prime`, so it can't be installed. That package cannot be recommended.

I would rather just state that _a_ script can be used to set these variables for convenience.

lennie420, 2021-07-06:
@MilesBHuff The prime-run command in Arch is just those commands in a script. See the following: https://github.com/archlinux/svntogit-packages/blob/packages/nvidia-prime/trunk/prime-run from https://archlinux.org/packages/extra/any/nvidia-prime/. I think talking about the AUR is kind of unrelated to Pop!_OS as well.

MilesBHuff, 2021-07-12:
@lennie420 
> @MilesBHuff The prime-run command in Arch is just those commands in a script.

Right -- I was under the impression that that's what `prime-run` was everywhere!  If it isn't, then maybe System76 could provide its own wrapper script with those variables.  It's certainly handy to have!

> I think talking about the AUR is kind of unrelated to Pop!_OS as well.

Fwiu, System76 actually maintains its own PKGBUILDs for the AUR, so while it's not related to Pop!_OS, it *is* related to system76-power.

crawfxrd, 2021-07-12:
We do not maintain anything related to Arch. Packages on AUR are not provided by us, and we do not provide the `PKGBUILD` for them.

MilesBHuff, 2021-07-12:
@crawfxrd 
> We do not maintain anything related to Arch. Packages on AUR are not maintained by us, and we do not provide the `PKGBUILD` for them.

Thanks for clarifying!  I must have misread somewhere.  I remember being surprised at the time, since I'd figured you'd only be supporting Pop!_OS.