pop-os/system76-power #248: Increased power draw in laptops after wakeup from suspension.

DuttaAB-dev, 2021-05-06:
**Distribution (run cat /etc/os-release):**
NAME="Pop!_OS"
VERSION="21.04"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 21.04"
VERSION_ID="21.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=hirsute
UBUNTU_CODENAME=hirsute
LOGO=distributor-logo-pop-os

neofetch:

[image]

 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**

system76-power:
  Installed: 1.1.16~1620080479~21.04~b74bcb7
  Candidate: 1.1.16~1620080479~21.04~b74bcb7
  Version table:
 *** 1.1.16~1620080479~21.04~b74bcb7 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu hirsute/main amd64 Packages
        100 /var/lib/dpkg/status

**Issue/Bug Description:**
The laptop runs normally after booting, consumes only 4-6W of power under normal usage(integrated graphics, battery life mode) and continues so even after using it for hours. But after waking it up from sleep, power draw increases to 10-15W even in idle. Also CPU usage increases dramatically.
This remains until the device is rebooted. And it seems that the reason behind this is that the nvidia gpu gets enabled after waking the system up.

Before suspending the laptop-

[image]

neofetch:

[image]

After waking up from suspension:-

[image]

neofetch:

[image]
[Note the added nvidia gpu in the system info.]

**Steps to reproduce :**
Set the laptop to sleep/turn the lid off and then wake again.

**Expected behavior:**
Normal power usage(4-6W)

**Other notes**
I had opened a similar issue in the pop-os repo last year #1156

bflanagin, 2021-05-07:
@DuttaAB-dev , what are your current power settings so I can try to replicate the issue.

DuttaAB-dev, 2021-05-07:
Battery saver

DuttaAB-dev, 2021-06-12:
> @DuttaAB-dev , what are your current power settings so I can try to replicate the issue.

Were you able to replicate this issue?