pop-os/system76-power #250: Battery percentage sometimes won't show

davidFagerburg, 2021-05-14:
**Distribution (run `cat /etc/os-release`):**
 

>  NAME="Pop!_OS"
> VERSION="20.10"
> ID=pop
> ID_LIKE="ubuntu debian"
> PRETTY_NAME="Pop!_OS 20.10"
> VERSION_ID="20.10"
> HOME_URL="https://pop.system76.com"
> SUPPORT_URL="https://support.system76.com"
> BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
> PRIVACY_POLICY_URL="https://system76.com/privacy"
> VERSION_CODENAME=groovy
> UBUNTU_CODENAME=groovy
> LOGO=distributor-logo-pop-os

 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**

Not sure what package it would be

**Issue/Bug Description:**

Occaisionally the top right dropdown menu appears to become the default ubuntu drop down menu, not the pop os one. It doesn't show the option to switch between compute, hybrid, or nvidia graphics like it normally does

**Steps to reproduce (if you know):**

I don't know. I recently installed updated to pop os

**Expected behavior:**

The drop down menu consistently uses the same format, showing the pop os options to switch graphics.

**Other Notes:**

bflanagin, 2021-05-14:
Renamed and moved. 

@davidFagerburg thanks for your report, do you have any non-default gnome extensions installed or packages like laptop-mode-tools?

vycoder, 2021-07-20:
I'm not sure if it's related but I've been having USD read issues for a couple of weeks now and I noticed that the power options label also disappeared the same time this is happening.

I'm using an XPS 15 9570 and sometime a few weeks ago, can't track which update it was after. USB ports stopped working after boot. I can't external keyboards, mice, anything for 5 to 15minutes or so after initial boot.

It doesn't seem to be a hardware ports issue, since I can use the external keyboard just fine when Pop OS prompts me for the disk decryption password. 

I've upgraded to a fresh install of 20.04 when it was first released but the problem is still there. The wait time was greatly reduced though, USB now works in about 5 minutes. Also, there's no Battery Icon, no Battery Indicator, it's now just a dropdown arrow icon. Power Options label is blank as well. There's a chevron icon, which brings up the power options but no label.

```
NAME="Pop!_OS"
VERSION="21.04"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 21.04"
VERSION_ID="21.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=hirsute
UBUNTU_CODENAME=hirsute
LOGO=distributor-logo-pop-os
```

I'm also seeing the following error on `dmesg`
```
[18185.486962] usb 1-12: new high-speed USB device number 70 using xhci_hcd
[18190.750930] usb 1-12: device descriptor read/64, error -71
[18195.878866] usb 1-12: device descriptor read/64, error -71
[18195.986773] usb usb1-port12: attempt power cycle
```
That goes on indefinitely starting from *new high-speed USB device number 7 using xhci_hcd* up to 127 and goes back to 7 again. The laptop is running considerably hot as well, been having this issue for quite some time now.

This might be a dell firmware issue but it seems like it's related to system-76 power.

kruthoff, 2022-01-28:
I observe the same behavior since today. The power indicator icon has vanished (if running on battery or charging). The menu to switch the power profile has disappeared, too. The power profile can be switched on the cmdline, but the desktop doesn't show it anymore. :-(

I don't use additional gnome extensions. Just vanilla Pop_OS! 21.10 on a system76 gazelle 14.

 Gnome 40.4.0. Kernel 5.15.15-76051515-generic #202201160435~1642693824~21.10~97db1bb SMP Thu Jan 20 17:35:05 U x86_64 x86_64 x86_64 GNU/Linux

I've logged out, logged in again, rebooted, logged in with another user, it didn't help. The options are gone. Any help would be much appreciated.

rfry85, 2023-01-13:
The power drop down entry dissapears when my laptops are plugged in and reappears when running on battery also. I have this I have a Lenova and system76 running POP OS 22.04 using this extention.

Soezdemir23, 2023-11-13:
Hello, I was trying to find a solution to this problem and found this place instead.

I also ran cat /etc/os-release and have the following information:

> NAME="Pop!_OS"
> VERSION="22.04 LTS"
> ID=pop
> ID_LIKE="ubuntu debian"
> PRETTY_NAME="Pop!_OS 22.04 LTS"
> VERSION_ID="22.04"
> HOME_URL="https://pop.system76.com"
> SUPPORT_URL="https://support.system76.com"
> BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
> PRIVACY_POLICY_URL="https://system76.com/privacy"
> VERSION_CODENAME=jammy
> UBUNTU_CODENAME=jammy
> LOGO=distributor-logo-pop-os

Kernel version is:
> 6.5.6-76060506-generic

It is infuriating since I have to shutdown and the turn on the notebook (Asus strix g513qm) several times before it sticks to the nvidia graphics.