pop-os/system76-power #252: Set graphics to integrated only doesn't work, it keeps GPU available on Compute mode.

mende1, 2021-05-27:
The difference between Compute mode and Integrated mode is that one keep GPU on and available and other completely shutdown GPU.

When I set:
`sudo system76-power graphics integrated`

And then I check:
`sudo system76-power graphics`

It return `compute`, and I want tthe `integrated` mode to keep some charge on my laptop.
Is there a package that I forgot to install? How can I fix this?
Thanks.

Some info:
I use arch, btw
And [system76-power](https://aur.archlinux.org/packages/system76-power/) package is that from AUR repository.

AnInternetTroll, 2021-06-26:
Hi, I am having the same issue on a fresh arch install.

Masteroduo, 2021-12-05:
Hi did you manage to fix it?

Techseeker-404, 2021-12-16:
@mende1 Did you reboot it after setting it to integrated graphics?!.The GPU mode switching will only take effect after you reboot the system. Anyways compute mode will not drain that much charge of battery as like in hybrid mode . Also changing mode and rebooting the system will cost much of battery power.

Mango-D, 2022-03-05:
Hi, I am having the same issue, Is there a fix yet?
I am also on arch.

mmstick, 2022-03-05:
You have to regenerate your initramfs

CygnusTerminal, 2022-03-21:
Chiming in to say i am having issues too.

mmstick, 2022-03-22:
You'll have to report that the Arch maintainer

CygnusTerminal, 2022-03-22:
> You'll have to report that the Arch maintainer

Thanks, i already did, hope they fix this issue, this is the best GPU/power manager, bar none.

ericpubu, 2022-04-11:
I was dealing with this issue for a while, until I had sometime these days to take a deeper look into the code to see if there was something failing with the GPU change here, or if maybe arch needed some other patch besides the change from `update-initramfs -u` to `mkinitcpio -P` to regenerate the intramfs (that the AUR package already have) but I found out that `system-76-power` was doing everything correctly.
So it had to be something else with my system, and after hours of search, I found out that something else was loading the nvidia module, and in my case it was another module `nvidia_uvm` and It didn't allow unloading it on a graphical session, I could unload it on a tty, but after a reboot, it was loaded again. It turns out that the `nvidia-utils` package was creating a file in `/etc/modules-load.d` that loaded this module on boot every time, after deleting it, finally I could see the graphics on just integrated and the nvidia module unloaded. And I can switch on and off and use nvidia perfectly without that `nvidia_uvm` module.
This may not be the problem for everyone, but it is almost certain that is some other package loading the nvidia package at boot without you noticing it, I thought maybe sharing it can help someone else here.

TL;DR: It's almost certain that the issue on most arch users, is another package loading the module at boot, try looking into `/etc/modules-load.d`, `etc/modprobe.d` or `usr/lib/modules-load.d` if anything is loading the nvidia module or something that depends on it.

closed 2025-09-17 by crawfxrd