pop-os/system76-power #255: More flexible make install paths

tleydxdy, 2021-06-20:
**Distribution (run `cat /etc/os-release`):**
 Arch Linux

**Issue/Bug Description:**
When building the package for Arch, I found out that Arch want the dbus configuration in `/usr/share/dbus-1/system.d/system76-power.conf` and the systemd service file in `/usr/lib/systemd/system/system76-power.service` however the current makefile can't be made to does that https://github.com/pop-os/system76-power/blob/master/Makefile#L38

So I wonder if this can be accommodated somehow. I'm not sure what's the difference between `/etc/dbus-1` and `/usr/share/dbus-1` on debian-famliy

Thanks.

crawfxrd, 2021-06-20:
Looking at every other systemd service file on a system, it should be installed to `$(libdir)` and a symlink added in `$(sysconfdir)`.

tleydxdy, 2021-06-21:
Per my understanding /etc should be where user added/edited config goes, and /usr/lib and /usr/share are where package installed configs should go. I do see that pop-os seems to do things differently for some packages but not all, I don't know why.

closed 2022-04-18 by mmstick