pop-os/system76-power #262: Can't enable the extension

mancarv, 2021-07-16:
**Distribution (run `cat /etc/os-release`):**
 NAME="Pop!_OS"
VERSION="20.04 LTS"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 20.04 LTS"
VERSION_ID="20.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
LOGO=distributor-logo-pop-os
 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
system76-power:
  Instalados: 1.1.17~1626197048~20.04~5a46c86
  Candidato:  1.1.17~1626197048~20.04~5a46c86
  Tabla de versión:
 *** 1.1.17~1626197048~20.04~5a46c86 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu focal/main amd64 Packages
        100 /var/lib/dpkg/status

**Issue/Bug Description:**
I used extensions.gnome.org/local to disable the utility and now I can't enable again. It doesn't show any errors at first sight.

**Steps to reproduce (if you know):**

1. Open https://extensions.gnome.org/local/
2. Find the extension 
[image]
3. Try to switch on
4. It looks like it worked but the the options to change the performance profile still doesn't show up
5. If I watch the network tab in the dev tools of the browser I see this
[image]
[image]

**Expected behavior:**
This is how it looks when I switch on/off any other extension

**Other Notes:**
It looks like it fail when any extension of System76 :(

jacobgkau, 2021-07-16:
Can you please try using the Extensions app (green icon) instead of the webpage?

mancarv, 2021-07-19:
It worked 🤦🏼‍♂️ thanks! I have a last question, if I disable the extension the configuration is removed? I mean if I set "Battery Life" and disable the extension this profile will be applied forever o it come back to whatever it was before I set any profile?

jacobgkau, 2021-07-19:
The extension is just a frontend for the daemon. Disabling the extension would not change the mode that the daemon is set to. However, even with the extension installed, the selected power profile is not saved between reboots; to have a power profile persist, you could set up a systemd service or something similar to run `system76-power profile battery` on boot, after the system76-power service has started.

closed 2021-07-19 by jacobgkau