pop-os/system76-power #263: Performance Mode Setting does not persist

Geigerkind, 2021-07-30:
**Distribution (run `cat /etc/os-release`):**
```
NAME="Pop!_OS"
VERSION="21.04"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 21.04"
VERSION_ID="21.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=hirsute
UBUNTU_CODENAME=hirsute
LOGO=distributor-logo-pop-os
```

**Issue/Bug Description:**
I always set my laptop to "High Performance Mode" but the change does not persist, it always goes back to balanced.

**Steps to reproduce (if you know):**
Havent found out what causes this. But usually I need to start the day with putting it to performance mode. Maybe a restart?

**Expected behavior:**
Mode setting should persist.

jacobgkau, 2021-07-30:
In the past, some of these have been closed as duplicates of https://github.com/pop-os/system76-power/issues/2, since config files would imply the ability to save/load the selected power profile; but since that issue doesn't mention persistable power profiles specifically (it's more about tweaking what the profiles are doing), I'll leave this one open unless anyone disagrees.

It looks like there's a PR that attempted to implement this feature (https://github.com/pop-os/system76-power/pull/88), although it's not building on any current distribution versions, so it would need to be updated or replaced.

hallamjeff, 2021-08-05:
I have the same issue as the OP on a Galago Pro 5. How are these power profiles related to the charge thresholds set on the CLI (if at all)?

jacobgkau, 2021-08-05:
> How are these power profiles related to the charge thresholds set on the CLI (if at all)?

Charge thresholds, power profiles, and graphics profiles are all separate settings that are not related; only graphics profiles are saved between reboots.

There's a pending documentation PR to add instructions to set charge profiles at boot (which would also work for power profiles, since it's just automating running the command at boot): https://github.com/system76/docs/pull/584/files

hallamjeff, 2021-08-05:
Excellent, I've put that to use now.

Is naming the power profile "Battery Life" potentially misleading that it relates to the lifespan of the battery (which has more to do with charge profiles?) Just speculating that a power profile name such as "Endurance" or "Conserve Battery" may be more clear.

jacobgkau, 2021-08-05:
> Is naming the power profile "Battery Life" potentially misleading that it relates to the lifespan of the battery (which has more to do with charge profiles?) Just speculating that a power profile name such as "Endurance" or "Conserve Battery" may be more clear.

I think the term "battery life" is generally understood to mean how long your battery will last on its current charge. I could just as easily say that "endurance" sounds more like long-term endurance over its lifespan than short-term charge. It's an interesting consideration, though.

curiousercreative, 2022-04-12:
[This systemd service solution](https://gist.github.com/curiousercreative/09aa4ccb81b45ce46f2fc063abf5b6cf seems to work FWIW. Feels like a bit of an arms race where I now have a default governor set in kernel boot option, then system76-power overrides it, and then finally (hopefully) this service overrides it yet again.