pop-os/system76-power #264: Charge Thresholds ignored when docked

hallamjeff, 2021-08-05:
**Distribution (run `cat /etc/os-release`):**
 
```
 NAME="Pop!_OS"
VERSION="21.04"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 21.04"
VERSION_ID="21.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=hirsute
UBUNTU_CODENAME=hirsute
LOGO=distributor-logo-pop-os
```
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**

```
system76-power:
  Installed: 1.1.17~1626370664~21.04~82bf0f4
  Candidate: 1.1.17~1626370664~21.04~82bf0f4
  Version table:
 *** 1.1.17~1626370664~21.04~82bf0f4 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu hirsute/main amd64 Packages
        100 /var/lib/dpkg/status
```

**Issue/Bug Description:**

Regardless of whether I set the power charge thresholds when the system is up or upon init (using the currently proposed systemd unit) my **docked** Galago Pro 5 doesn't dip below 100% charge. I can see that battery lifetime is already at 94.3% of manufactured capacity (possibly as a result).

Output of `system76-power charge-thresholds` :

```
Profile: Custom
Start: 45
End: 55
```
systemd unit file:
```

[Unit]
Description=Set the charge threshold at startup.
After=default.target

[Service]
Type=simple
ExecStart=system76-power charge-thresholds 45 55

[Install]
WantedBy=default.target
```

**Steps to reproduce (if you know):**

Dock a fully charged Galago Pro 5 and attempt to set the charge thresholds to the parameters listed above.

**Expected behavior:**

Galago Pro 5 should discharge until reaching the low-end charge threshold and then maintain between the two thresholds, even while docked.

**Other Notes:**

Battery info as listed by "Power Statistics" app:

[image]

crawfxrd, 2021-08-05:
It does discharge. You just have nothing drawing power from the battery because you're powering it via AC/USB, so it will take a very long time.

hallamjeff, 2021-08-05:
That makes sense I suppose I interpreted charging to also mean running on AC but you're saying those are distinct possibilities. For life of the battery I guess I should undock it and run it down to 50% or so before docking it again. I suppose there's no way for it to run on battery while docked just to have the external displays and keyboard etc.?

crawfxrd, 2021-08-05:
Correct. If AC/USB power is present it will always be used over the battery. (This is controlled by the [EC firmware](https://github.com/system76/ec/blob/e00686a6615af40ffacb45a744e7f1da2e49154d/src/board/system76/common/power.c#L424-L462).) So you should run it off battery to whatever value you want before plugging it in.

hallamjeff, 2021-08-05:
Closing because my premise was incorrect. Thank you for the feedback.

hallamjeff, 2021-08-08:
Re-opening the issue just to seek clarification for one other point: The charge thresholds were respected after I undocked and depleted the battery until below the upper cutoff. However, if I power off the laptop at the end of the day (while still docked) it loses this OS-level guidance and charges back to 100%, and of course remains there unless I undock it and run the battery down again.

Is it possible, and is there any plan, to somehow maintain the thresholds when powered off? Does everyone else leave their docked laptops on 24/7 or am I missing something?

crawfxrd, 2021-08-08:
system76/ec#128

If the EC resets, then these values also reset to 0 100.

Does the AC LED turn on when connected to the dock? If so, does it turn off at all when you power off the laptop?

hallamjeff, 2021-08-08:
The AC LED turns on when connected to the dock and when I power off the laptop the AC LED turns orange.

trevor87, 2023-05-20:
I have the same problem as described below on my lemp10 using Pop!_OS 22.04 LTS with [the workaround](https://support.system76.com/articles/laptop-battery-thresholds/#at-boot) and the Maximum Lifespan Profile. Since it sometimes works and sometimes doesn't I had trouble figuring out what is happening. Here is what I found:

When using the original laptop charger everything works: When the laptop is plugged in and above my charging threshold and I turn if off, the thresholds are respected even while it is off. I checked and the charging light instantly switches from green to orange on poweroff.

When using my USB-C dock, the following happens:

- First the laptop is not plugged in
- Then I plug it in to my USB-C dock, the threshold is respected while the laptop is on
- Then I  power it off. While doing this I notice that the charging light does not instantly switch to green, but briefly turns off and then turns orange.
- The laptop now continues to charge even though it shouldn't as it is above 50%.
- When turning the laptop on again the thresholds are respected again.
- And now the curious part: When I turn if off the second time, without having unplugged the USB-C, the charging light instantly switches to green and the thresholds are preserved. But weird things happen: I do not get any image on my external monitor and boot time is a lot longer than after the first power off (immediately after plugging in the USB-C dock).

This behavior is also reproducible.

> Re-opening the issue just to seek clarification for one other point: The charge thresholds were respected after I undocked and depleted the battery until below the upper cutoff. However, if I power [...]