pop-os/system76-power #265: Can't set power profile

CSpencerND, 2021-08-16:
**Distribution (run `cat /etc/os-release`):**
```
NAME=ArcoLinux
ID=arcolinux
ID_LIKE=arch
BUILD_ID=rolling
ANSI_COLOR="0;36"
HOME_URL="https://arcolinux.info/"
SUPPORT_URL="https://arcolinuxforum.com/"
BUG_REPORT_URL="https://github.com/arcolinux"
LOGO=arcolinux-hello
IMAGE_ID=arcolinuxb-qtile
IMAGE_VERSION=v21.07.2
 ```

**Issue/Bug Description:**
Error received after attempting to set a power profile
```
setting power profile to Performance
daemon returned an error message: "Errors found when setting profile:
    - failed to set scsi host profiles: failed to set link time power management policy med_power_with_dipm on host0: No such file or directory (os error 2)"
```

**Steps to reproduce (if you know):**
`system76-power profile performance` or `balanced` or `battery`

**Expected behavior:**
Set a power profile

**Other Notes:**
Charge Thresholds works fine.
Darter Pro 7

CSpencerND, 2021-08-16:
### Update
I just tried to set a profile again and it seems to have worked (I got `setting power profile` returned and the brightness changed). Then I tried to switch to another profile and I got the error message posted above. Tried again, and sometimes it says it worked but the brightness doesn't change. 

It seems random (to me) which of the three possible outcomes is produced:

- `failed to set scsi host profiles...`
- `setting power profile...` is returned but nothing seems to happen
- `setting power profile...` and I see the brightness change

_I'm using the brightness as a gauge of whether or not it worked because I have no idea how else I could know if it's doing anything._

CSpencerND, 2021-09-03:
### Update
I got it working every time guaranteed with this function:
```
s76-profile () {
until system76-power profile $1; do
  sleep .2
done
}
```

nnleaf, 2021-09-11:
Hi, I'm running Arch and tried installing system76-power from AUR. 

When I run `system76-power` , I get this : 
```
Power Profile: ?
```

I've tried setting it manually with `system76-power profile battery` it will output this :
```
setting power profile to Battery
daemon returned an error message: "Errors found when setting profile:
    - failed to set scsi host profiles: failed to set link time power management policy min_power on host0: No such file or directory (os error 2)"
```
That seems to be the exact error you got. If I run the command a few more times, it will eventually switch profiles.
```
Power Profile: Battery
CPU: 10% - 50%, No Turbo
Backlight intel_backlight: 106/1060 = 10%
Keyboard Backlight tpacpi::kbd_backlight: 0/2 = 0%
```
Trying to switch it back to performance profile, I'll get the same error, have to run the command a few times before the error is gone and the profile is set. 

Setting between profiles, the brightness changes and the CPU % ranges changes (Battery = 10-50%, balanced = 10-100%, performance = 50-100%) based on what `system76-power profile` reports back. Also according to `powertop`, it seems to work as my discharge goes down slightly on the battery profile (Tested with the screen brightness on all profiles. 

Did you ever get it to reliably switch profiles? Having to run the command 2-3 times before it sticks is quite annoying. The gnome-extension doesn't work either.

CSpencerND, 2021-09-11:
I wasn't able to fix it but I wrote a rofi script to make a gui for it with an attempt limit , that also `notify-send`s the profile status whether it works or not.
```
#!/usr/bin/env bash

options=$(echo "Battery
Balanced
Performance" | rofi -dmenu -p "Power Profile")

[[ -z $options ]] && notify-send "$(system76-power profile)" && exit

s76-run () {
    x=0
    until [ "$x" -ge 3 ]; do
      system76-power profile $1 && break
      x=$((x+1))
      sleep .3
    done

  notify-send "$(system76-power profile)"
}

[[ $options = *Bat* ]] && s76-run battery
[[ $options = *Bal* ]] && s76-run balanced
[[ $options = *Per* ]] && s76-run performance
```

eduncan911, 2022-01-04:
This isn't isolated to just Arch/ArcoLinux.  I wanted to confirm the same ever since I installed Pop_OS (Ubuntu) 18.04 (2018, Apr).  I've been waiting years for it to be fixed.

> $ system76-power profile battery
> setting power profile to Battery
> daemon returned an error message: "Errors found when setting profile:
>     failed to set scsi host profiles: failed to set link time power management policy min_power on host0: No such file or directory (os error 2)

For the record, I thought it was my odd-ball installation:

* Pop_OS 18.04 -> 18.10 -> 19.04 -> 19.10 -> 20.04 -> 21.10 (2021, Oct)
* Installed Regolith-Desktop around 19.04 (i3-wm + i3-gaps + i3-gnome-session)

Nice to see it's not just me.

jacobgkau, 2022-01-04:
@eduncan911 What hardware are you using? This application works on both currently-supported Pop!_OS releases on all System76 hardware. Newer kernels that do not expose the same files as old ones, newer GNOME, or hardware that does not support all of the features that the app tries to set would be likely reasons for it not working.

For the `min_power` issue, perhaps this line needs to be wrapped in an "if" statement like some of the ones later in the file: https://github.com/pop-os/system76-power/blob/516668290488c628e64b954bd0f9fe192742126a/src/daemon/profiles.rs#L116 

For the `med_power_with_dipm` issue (which is what was initially reported in this GitHub issue), it looks like it would be these two lines: https://github.com/pop-os/system76-power/blob/94b057eb39fdc82b68c8940b592bc0015c2ed06a/src/daemon/profiles.rs#L55 & https://github.com/pop-os/system76-power/blob/94b057eb39fdc82b68c8940b592bc0015c2ed06a/src/daemon/profiles.rs#L93

eduncan911, 2022-01-04:
@jacobgkau thanks for looking into this!

I have a Thinkpad P1, Gen 1.  Been getting this error when setting profiles ever since I installed Pop on the machine when it was new.

And just like others, the first time it failed, the second it always succeeds in my case though.

Ddystopia, 2023-01-24:
Darter Pro, 
`uname -a`
`Linux ddps76 6.1.7-arch1-1 #1 SMP PREEMPT_DYNAMIC Wed, 18 Jan 2023 19:54:38 +0000 x86_64 GNU/Linux`

Error:
`setting power profile to Battery
daemon returned an error message: "Errors found when setting profile:
    - failed to set scsi host profiles: failed to set link time power management policy min_power on host0: No such file or directory (os error 2)"
`

Conditions are the same as mentioned

lpingree, 2023-12-06:
Any new status on this?

mmstick, 2023-12-07:
That's an acceptable warning. The system is able to successfully set power profiles regardless of its ability to control the power management policy of SATA drives.