pop-os/system76-power #272: Documentation for differences between profiles

31337-4554551n, 2021-09-21:
I would like to request some documentation in the differences between battery life, balanced, and high performance profiles for the readme please.
It's very difficult (and I still haven't been able to) to get information on what the actual changes made by these profiles are, with the exception of brightness which is obvious when they are selected.
Also, if/how they interact/interfere with tlp/powertop

jacobgkau, 2021-09-21:
> It's very difficult (and I still haven't been able to) to get information on what the actual changes made by these profiles are, with the exception of brightness which is obvious when they are selected.

Not saying this shouldn't be documented in the readme, but the profiles seem to be defined in `profiles.rs`: https://github.com/pop-os/system76-power/blob/master/src/daemon/profiles.rs 

The `balanced` function in particular has comments explaining each of the items being set.

31337-4554551n, 2021-09-23:
The comments there really don't provide enough information in terms on what's actually going on, the high performance isn't documented at all.
It would be really nice to have a summary of how they differ from each other.