pop-os/system76-power #277: system76-power daemon is taking too much cpu

edap, 2021-10-03:
Previously open on https://github.com/pop-os/system76-firmware/issues/68

**Distribution (run `cat /etc/os-release`):**

NAME="Pop!_OS"
VERSION="21.04"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 21.04"
VERSION_ID="21.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=hirsute
UBUNTU_CODENAME=hirsute
LOGO=distributor-logo-pop-os

 
 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**

System 76 power daemon

**Issue/Bug Description:**

Taking 100% cpu

**Steps to reproduce (if you know):**

I do not know. But one processor is always using 100% of the resources, independently from which application is running

**Expected behavior:**

When just the browser is running (without videos webGL applications), the daemon should not take that amount of cpu and the fan should be quiet.

**Other Notes:**
[image]

crawfxrd, 2021-10-04:
How frequently does this happen?

What does systemctl say about the service when it's in that state?

```
sudo systemctl status system76-power
```

grantbrown, 2021-10-04:
This occurs about every other week for me. I also sometimes observe that this daemon hangs, preventing proper shutdown. 

Here's the result of running that command from this morning (ended up here because it happened just now and wondered if anyone had a solution).

```
 system76-power.service - System76 Power Daemon
     Loaded: loaded (/etc/systemd/system/system76-power.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2021-10-04 08:35:25 CDT; 2h 29min ago
   Main PID: 4427 (system76-power)
      Tasks: 18 (limit: 47308)
     Memory: 5.3M
     CGroup: /system.slice/system76-power.service
             ├─ 812 /usr/bin/system76-power daemon
             └─4427 /usr/bin/system76-power daemon

Oct 04 08:35:25 zg14 system76-power[4427]: [INFO] Disabling NMI Watchdog (for kernel debugging only)
Oct 04 08:35:25 zg14 system76-power[4427]: [INFO] Setting automatic graphics power
Oct 04 08:35:25 zg14 system76-power[4427]: [INFO] Enabling graphics power
Oct 04 08:35:25 zg14 system76-power[4427]: [INFO] Initializing with the balanced profile
Oct 04 08:35:25 zg14 system76-power[4427]: [INFO] Registering dbus name com.system76.PowerDaemon
Oct 04 08:35:25 zg14 system76-power[4427]: [INFO] Adding dbus path /com/system76/PowerDaemon with interface com.system76.PowerDaemon
Oct 04 08:35:25 zg14 systemd[1]: Started System76 Power Daemon.
Oct 04 08:35:25 zg14 system76-power[4427]: [ERROR] fan daemon: platform hwmon not found
Oct 04 08:35:25 zg14 system76-power[4427]: [INFO] Handling dbus requests
Oct 04 08:35:25 zg14 system76-power[4427]: [ERROR] hid_backlight: no system76_acpi::kbd_backlight led

```
Here's the result of `cat /etc/os-release` for me:

````
NAME="Pop!_OS"
VERSION="21.04"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 21.04"
VERSION_ID="21.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=hirsute
UBUNTU_CODENAME=hirsute
LOGO=distributor-logo-pop-os
````

Also, I am not on System76 hardware if it matters - I'm observing this on a Rog Zephyrus G14 (GA401IV-BR9N6)

edap, 2021-10-04:
I am also on a G14, (ROG Zephyrus G14 GA401IV-HE003, processor AMD R7 4800HS, graphics card RTX 2060). Occasionally I also have troubles shutting down the laptop. I will post the result of `sudo systemctl status system76-power` as soon as the problem happens again.

crawfxrd, 2021-10-04:
This looks like it may be the same issue as #268.

crawfxrd, 2021-10-04:
You can also try installing strace. Then, once you hit the issue, if you run `htop` as root you can press `s` (and then F8 for autoscroll) to see what syscall it's spinning on.

grantbrown, 2021-10-04:
Thanks, will try that when it recurs.

thazhemadam, 2021-11-01:
This seems to be happening to me too, although it is a _**lot**_ more frequent than once a week - more like at least once every two days. Is there any fix/hack around that could help fix this? Even sending a `SIGKILL` to the process doesn't seem to kill the process.

emptythevoid, 2023-03-12:
Is the same as this bug?  https://github.com/pop-os/system76-power/issues/360