pop-os/system76-power #280: Error on `system76-power graphics power auto`

jacobgkau, 2021-10-08:
**Distribution (run `cat /etc/os-release`):** Pop!_OS 21.04 and 20.04 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):** efd0754 (current `master` branch)

**Issue/Bug Description:** When a system is in Compute graphics mode, the discrete graphics can be turned off and on using `system76-power graphics power off` and `system76-power graphics power on`. The help text for `system76-power graphics power` shows that a third setting, `auto`, is available, but when attempting to switch to that mode, I get the following output:

```
setting discrete graphics to turn off when not in use
daemon returned an error message: "Unknown method AutoGraphicsPower"
```

This is different from the message shown when a nonexistent mode is specified.

**Steps to reproduce (if you know):** Run `system76-power graphics power auto` on any system (switchable or non-switchable.)

**Expected behavior:** Either the command should complete successfully, or it should be removed/hidden from the help text.

**Other Notes:** N/A

crawfxrd, 2021-10-08:
Caused by: 065312d5c6f (#195)

Should just need to add `AutoGraphicsPower` back to the dbus interface.

curiousercreative, 2023-02-06:
@jacobgkau I've got this compiling now, but maybe there's more work to do as my galp5 dGPU power is set to on by this when I'm in integrated graphics mode whereas I expect it to be off.

```
$ system76-power graphics
integrated
$ system76-power graphics power
off (discrete)
$ system76-power graphics power auto
setting discrete graphics to turn off when not in use
$ system76-power graphics power
on (discrete)
$ system76-power graphics power off
turning discrete graphics off 
$ system76-power graphics power
off (discrete)
$ system76-power graphics power auto
setting discrete graphics to turn off when not in use
$ system76-power graphics power
on (discrete)
```

closed 2025-10-24 by jacobgkau