pop-os/system76-power #289: [Query] How to add system76 PPA on Ubuntu 20.04.3 and pin only the system76-power package ?

tur1ngb0x, 2021-11-04:
**Distribution (run `cat /etc/os-release`):**
 NAME="Ubuntu"
VERSION="20.04.3 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.3 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal

**Issue:**
I want to add this PPA `system76-dev/stable` and just install `system76-power` package and nothing else.
When I update the Ubuntu system, other Ubuntu packages get updated to PopOS packages which I want to avoid.

**Steps to reproduce (if you know):**
`sudo apt-add-repository -ys ppa:system76-dev/stable` 
`sudo apt update`
`sudo apt install system76-power`
`sudo apt upgrade`
Other Ubuntu packages get updated from system76 repository

**Expected behavior:**
How can I set the priority of `system76-power` package the highest and the priority of all other packages from system76 repository lower than than Ubuntu repositories ?

jacobgkau, 2021-11-04:
What is the use case for this? The `system76-dev/stable` PPA is intended for Ubuntu, not Pop!_OS. Pop!_OS uses the `system76/pop` PPA.

This question is more about apt than system76-power, but you can read about pinning syntax here:
https://wiki.debian.org/AptConfiguration 
https://help.ubuntu.com/community/PinningHowto

tur1ngb0x, 2021-11-04:
that worked! you can close this issue :)

jacobgkau, 2021-11-04:
Please keep in mind that GitHub is not a support forum. For future questions that are not bug reports or related to development, the [Pop!_Chat](https://chat.pop-os.org) would be a great place to ask.

closed 2021-11-04 by jacobgkau