pop-os/system76-power #294: Application does not work with GNOME 41

ahoneybun, 2021-11-30:
**Distribution (run `cat /etc/os-release`):**
 
NAME="Arch Linux"
PRETTY_NAME="Arch Linux"
ID=arch
BUILD_ID=rolling
ANSI_COLOR="38;2;23;147;209"
HOME_URL="https://archlinux.org/"
DOCUMENTATION_URL="https://wiki.archlinux.org/"
SUPPORT_URL="https://bbs.archlinux.org/"
BUG_REPORT_URL="https://bugs.archlinux.org/"
LOGO=archlinux-logo

**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**

```shell
pacman -Qs system76-power
local/gnome-shell-extension-system76-power-git 2.0.2.r0.ga901dc0-1
    Gnome shell extension for System76 power management
local/system76-power 1.1.19-1
    System76 Power Management
```

**Issue/Bug Description:**

Not able to use the `system76-power` package in Arch with GNOME 41 and it seems to happen in Fedora 35 as well.

```shell
systemctl status system76-power
○ system76-power.service - System76 Power Daemon
     Loaded: loaded (/usr/lib/systemd/system/system76-power.service; enabled; vendor preset: disabled)
     Active: inactive (dead) since Tue 2021-11-30 09:07:51 -03; 18min ago
    Process: 497 ExecStart=/usr/bin/system76-power daemon (code=exited, status=0/SUCCESS)
   Main PID: 497 (code=exited, status=0/SUCCESS)
        CPU: 19ms

Nov 30 09:07:49 galago system76-power[497]: [INFO] Adding dbus path /com/system76/PowerDaemon with interface com.system76.PowerDaemon
Nov 30 09:07:49 galago system76-power[497]: [ERROR] fan daemon: platform hwmon not found
Nov 30 09:07:49 galago system76-power[497]: [INFO] Handling dbus requests
Nov 30 09:07:49 galago systemd[1]: Started System76 Power Daemon.
Nov 30 09:07:49 galago system76-power[497]: [ERROR] hid_backlight: no system76_acpi::kbd_backlight led
Nov 30 09:07:51 galago systemd[1]: Stopping System76 Power Daemon...
Nov 30 09:07:51 galago system76-power[497]: [INFO] caught signal: SIGTERM
Nov 30 09:07:51 galago system76-power[497]: [INFO] daemon exited from loop
Nov 30 09:07:51 galago systemd[1]: system76-power.service: Deactivated successfully.
Nov 30 09:07:51 galago systemd[1]: Stopped System76 Power Daemon.
systemctl start system76-power
systemctl status system76-power
● system76-power.service - System76 Power Daemon
     Loaded: loaded (/usr/lib/systemd/system/system76-power.service; enabled; vendor preset: disabled)
     Active: active (running) since Tue 2021-11-30 09:29:11 -03; 1s ago
   Main PID: 8938 (system76-power)
      Tasks: 1 (limit: 38153)
     Memory: 828.0K
        CPU: 11ms
     CGroup: /system.slice/system76-power.service
             └─8938 /usr/bin/system76-power daemon

Nov 30 09:29:11 galago system76-power[8938]: [INFO] Disabling NMI Watchdog (for kernel debugging only)
Nov 30 09:29:11 galago system76-power[8938]: [INFO] Setting automatic graphics power
Nov 30 09:29:11 galago system76-power[8938]: [WARN] Failed to set automatic graphics power: does not have switchable graphics
Nov 30 09:29:11 galago system76-power[8938]: [INFO] Initializing with the balanced profile
Nov 30 09:29:11 galago system76-power[8938]: [INFO] Registering dbus name com.system76.PowerDaemon
Nov 30 09:29:11 galago system76-power[8938]: [INFO] Adding dbus path /com/system76/PowerDaemon with interface com.system76.PowerDaemon
Nov 30 09:29:11 galago systemd[1]: Started System76 Power Daemon.
Nov 30 09:29:11 galago system76-power[8938]: [ERROR] fan daemon: platform hwmon not found
Nov 30 09:29:11 galago system76-power[8938]: [INFO] Handling dbus requests
Nov 30 09:29:11 galago system76-power[8938]: [ERROR] hid_backlight: no system76_acpi::kbd_backlight led
```

**Steps to reproduce (if you know):**

1. Install `system-power` and `gnome-shell-extension-system76-power`
2. Enable and start systemd service
3. Use CLI to change the profile/Try to enable the extension 

**Expected behavior:**

Able to switch modes and profiles

**Other Notes:**

If you start the systemd service you are able to use the CLI but not the extension. This is the message you see when you try to use the CLI or enable the extension when you boot before starting the service:

```shell
Gio.DBusError: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name com.system76.PowerDaemon was not provided by any .service files
```

jacobgkau, 2021-11-30:
We have several issues open about Arch and Arch-based distros (and their kernel versions) already:
https://github.com/pop-os/system76-power/issues/276
https://github.com/pop-os/system76-power/issues/268
https://github.com/pop-os/system76-power/issues/265

If the issue being reported here is regarding the GNOME Shell extension, its repository is https://github.com/pop-os/gnome-shell-extension-system76-power.

ahoneybun, 2021-11-30:
I saw those other issues but the errors were different. I filed this against system76-power as the issue is with the main application not really the extension. There is a branch that I made for adding GNOME 41 to the metadata.json file.

mmstick, 2022-03-31:
Sounds like the Arch packaging is just outdated, and lacks the service file containing `BusName=com.system76.PowerDaemon`.

mmstick, 2022-03-31:
You could try installing system76-power from the `systemd` branch to see if this improves it any.

ahoneybun, 2022-03-31:
So running these commands are still needed as of testing today with GNOME 41.5:

```
sudo systemctl mask power-profiles-daemon
sudo systemctl start system76-power
```

I haven't been able to build the AUR package using the `systemd` branch yet.

kalmbach, 2022-04-18:
@ahoneybun  I had this issue in 22.04 (beta) 
```
"Gio.DBusError: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name com.system76.PowerDaemon was not provided by any .service files"
```
and was solved applying the same commands.
Thanks!