pop-os/system76-power #296: Can't switch from Integrated Graphics

binaryz3r0, 2021-12-07:
**Distribution (run `cat /etc/os-release`):**
DISTRIB_ID=Pop
DISTRIB_RELEASE=21.04
DISTRIB_CODENAME=hirsute
DISTRIB_DESCRIPTION="Pop!_OS 21.04"
NAME="Pop!_OS"
VERSION="21.04"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 21.04"
VERSION_ID="21.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=hirsute
UBUNTU_CODENAME=hirsute
LOGO=distributor-logo-pop-os
 

**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
system76-power:
  Instalado: 1.1.20~1637635456~21.04~b5ed171
  Candidato: 1.1.20~1637635456~21.04~b5ed171
  Tabela de Versão:
 *** 1.1.20~1637635456~21.04~b5ed171 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu hirsute/main amd64 Packages
        100 /var/lib/dpkg/status
     1.1.20~1637635456~21.04~b5ed171~dev 500
        500 http://ppa.launchpad.net/system76-dev/stable/ubuntu hirsute/main amd64 Packages

**Issue/Bug Description:**
I am unable to switch from Integrated Graphics. When I click any of the options (Hybrid Graphics, NVIDIA Graphics, Compute... etc) I get a message popup (for 30 seconds) staying that the change is being made, and then I am asked to reboot and switch. After reboot, Integrated Graphics is the only option selected.

**Steps to reproduce (if you know):**
I login to desktop, chose Hybrid graphics, reboot, but after reboot and login again to desktop, I see integrated graphics is selected again.

**Expected behavior:**
I login to desktop, chose hybrid graphics, reboot, and after reboot and login I see Hybrid Graphics selected.

**Other Notes:**
I've noticed this behaviour after apt upgrade from Pop_OS! 19 (or so I think) because this was working when I first installed Pop_OS!.
Its a dual boot machine, and on WIndows 10 side, everything is working. So its not a graphics card issue.
Graphic card is: NVIDIA Geforce GTX 1650 TI
Laptop: HP Pavilion Gaming 15S-ec1010np 

If I run lspci | grep -i VGA right before boot, I get: 
05:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Renoir (rev c7)

but if first, I run system76-power graphics power on and then lspci | grep -i VGA i get 
01:00.0 VGA compatible controller: NVIDIA Corporation TU117M [GeForce GTX 1650 Ti Mobile] (rev a1)
05:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Renoir (rev c7)

This shows that system76-power is disabling my discrete graphics card at boot. Why?

Thanks.
Regards,
Pedro

crawfxrd, 2021-12-07:
What's in the journal?

```
journalctl -b -u system76-power
```

When you switch to hybrid, is anything blacklisting the NVIDIA modules?

```
grep -r 'blacklist nvidia' /{etc,lib}/modprobe.d/
```

binaryz3r0, 2021-12-07:
journalctl -b -u system76-power output is:
dez 07 11:11:30 pavy systemd[1]: Starting System76 Power Daemon...
dez 07 11:11:30 pavy system76-power[754]: [INFO] Starting daemon
dez 07 11:11:30 pavy system76-power[754]: [INFO] Connecting to dbus system bus
dez 07 11:11:30 pavy system76-power[754]: [INFO] Rescanning PCI bus
dez 07 11:11:30 pavy system76-power[754]: [INFO] 0000:01:00.0: NVIDIA graphics
dez 07 11:11:30 pavy system76-power[754]: [INFO] 0000:01:00.0: Function for 0000:01:00.0
dez 07 11:11:30 pavy system76-power[754]: [INFO] 0000:01:00.1: Function for 0000:01:00.0
dez 07 11:11:30 pavy system76-power[754]: [INFO] 0000:05:00.0: AMD graphics
dez 07 11:11:30 pavy system76-power[754]: [INFO] 0000:05:00.3: Function for 0000:05:00.0
dez 07 11:11:30 pavy system76-power[754]: [INFO] 0000:05:00.6: Function for 0000:05:00.0
dez 07 11:11:30 pavy system76-power[754]: [INFO] 0000:05:00.4: Function for 0000:05:00.0
dez 07 11:11:30 pavy system76-power[754]: [INFO] 0000:05:00.2: Function for 0000:05:00.0
dez 07 11:11:30 pavy system76-power[754]: [INFO] 0000:05:00.0: Function for 0000:05:00.0
dez 07 11:11:30 pavy system76-power[754]: [INFO] 0000:05:00.5: Function for 0000:05:00.0
dez 07 11:11:30 pavy system76-power[754]: [INFO] Disabling NMI Watchdog (for kernel debugging only)
dez 07 11:11:30 pavy system76-power[754]: [INFO] Setting automatic graphics power
dez 07 11:11:30 pavy system76-power[754]: [INFO] Disabling graphics power
dez 07 11:11:30 pavy system76-power[754]: [INFO] 0000:01:00.0: Removing
dez 07 11:11:30 pavy system76-power[754]: [INFO] 0000:01:00.1: Removing
dez 07 11:11:31 pavy system76-power[754]: [INFO] Initializing with the balanced profile
dez 07 11:11:31 pavy system76-power[754]: [INFO] Registering dbus name com.system76.PowerDaemon
dez 07 11:11:31 pavy system76-power[754]: [INFO] Adding dbus path /com/system76/PowerDaemon with interface com.system76.PowerDaemon
dez 07 11:11:31 pavy systemd[1]: Started System76 Power Daemon.
dez 07 11:11:31 pavy system76-power[754]: [ERROR] fan daemon: platform hwmon not found
dez 07 11:11:31 pavy system76-power[754]: [INFO] Handling dbus requests
dez 07 11:11:31 pavy system76-power[754]: [ERROR] hid_backlight: no system76_acpi::kbd_backlight led
dez 07 11:11:54 pavy system76-power[754]: [INFO] DBUS Received GetSwitchable() method
dez 07 11:11:54 pavy system76-power[754]: [INFO] DBUS Received GetGraphics() method
dez 07 11:11:54 pavy system76-power[754]: [INFO] DBUS Received GetProfile() method
dez 07 14:30:23 pavy system76-power[754]: [INFO] DBUS Received GetSwitchable() method
dez 07 14:30:23 pavy system76-power[754]: [INFO] DBUS Received GetGraphics() method
dez 07 14:30:23 pavy system76-power[754]: [INFO] DBUS Received GetProfile() method

output of grep -r 'blacklist nvidia' /etc/modprobe.d/ : 
/etc/modprobe.d/blacklist-nouveau-nvidiafb.conf:blacklist nvidiafb

output of grep -r 'blacklist nvidia' /lib/modprobe.d/ : 
(empty)

Thanks

leviport, 2021-12-07:
My machine occasionally does this when either the Nvidia driver or the kernel is updated. I'm usually able to fix it by purging and reinstalling the Nvidia driver like so:

```
sudo apt purge ~nnvidia
sudo apt clean && sudo apt autoremove
sudo apt install nvidia-driver-470
```

binaryz3r0, 2021-12-14:
Hi leviport,

Thanks for your input. unfortunatelly. in my case that didn't work. Here's the bottom line of what happened:

`Loading new nvidia-470.86 DKMS files...
Building for 5.15.6-051506-generic
Building for architecture x86_64
Building initial module for 5.15.6-051506-generic
ERROR (dkms apport): kernel package linux-headers-5.15.6-051506-generic is not supported
Error! Bad return status for module build on kernel: 5.15.6-051506-generic (x86_64)
Consult /var/lib/dkms/nvidia/470.86/build/make.log for more information.
**dpkg**: erro ao processar o pacote nvidia-dkms-470 (--configure):
 o subprocesso instalado, do pacote nvidia-dkms-470, o script post-installation retornou erro do status de saída 10
**dpkg**: problemas com dependências impedem a configuração de nvidia-driver-470:
 nvidia-driver-470 depende de nvidia-dkms-470 (>= 470.86); no entanto:
 O pacote nvidia-dkms-470 ainda não está configurado.

**dpkg**: erro ao processar o pacote nvidia-driver-470 (--configure):
 problemas com dependências - a deixar por configurar
A instalar libnvidia-encode-470:amd64 (470.86-1pop0~1637635465~21.04~47710c4) ...
Nenhum relatório apport escrito pois a mensagem de erro indica que é um erro de seguimento de um erro anterior.
                                                                                                               A instalar libnvidia-encode-470:i386 (470.86-1pop0~1637635465~21.04~47710c4) ..
.
A processar 'triggers' para desktop-file-utils (0.26-1ubuntu1) ...
A processar 'triggers' para gnome-menus (3.36.0-1ubuntu1) ...
A processar 'triggers' para libc-bin (2.33-0ubuntu5) ...
A processar 'triggers' para man-db (2.9.4-2) ...
A processar 'triggers' para mailcap (3.68ubuntu1) ...
A processar 'triggers' para initramfs-tools (0.139ubuntu3) ...
update-initramfs: Generating /boot/initrd.img-5.15.6-051506-generic
W: Possible missing firmware /lib/firmware/amdgpu/yellow_carp_gpu_info.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/vangogh_gpu_info.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/aldebaran_ta.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/aldebaran_sos.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/aldebaran_rlc.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/aldebaran_mec2.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/aldebaran_mec.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/cyan_skillfish2_rlc.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/cyan_skillfish2_mec2.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/cyan_skillfish2_mec.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/cyan_skillfish2_me.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/cyan_skillfish2_pfp.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/cyan_skillfish2_ce.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/cyan_skillfish_rlc.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/cyan_skillfish_mec2.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/cyan_skillfish_mec.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/cyan_skillfish_me.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/cyan_skillfish_pfp.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/cyan_skillfish_ce.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/aldebaran_sdma.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/cyan_skillfish2_sdma1.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/cyan_skillfish2_sdma.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/cyan_skillfish_sdma1.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/cyan_skillfish_sdma.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/sienna_cichlid_mes.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi10_mes.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/aldebaran_vcn.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/aldebaran_smc.bin for module amdgpu
kernelstub.Config    : INFO     Looking for configuration...
kernelstub           : INFO     System information: 

    OS:..................Pop!_OS 21.04
    Root partition:....../dev/nvme0n1p7
    Root FS UUID:........e3ae3024-597a-4a5c-9f31-a0f1fe79ed34
    ESP Path:............/boot/efi
    ESP Partition:......./dev/nvme0n1p1
    ESP Partition #:.....1
    NVRAM entry #:.......-1
    Boot Variable #:.....0000
    Kernel Boot Options:.loglevel=0 splash systemd.show_status=false quiet mem_sleep_default=deep
    Kernel Image Path:.../boot/vmlinuz-5.15.6-051506-generic
    Initrd Image Path:.../boot/initrd.img-5.15.6-051506-generic
    Force-overwrite:.....False

kernelstub.Installer : INFO     Copying Kernel into ESP
kernelstub.Installer : INFO     Copying initrd.img into ESP
kernelstub.Installer : INFO     Setting up loader.conf configuration
kernelstub.Installer : INFO     Making entry file for Pop!_OS
kernelstub.Installer : INFO     Backing up old kernel
kernelstub.Installer : INFO     Making entry file for Pop!_OS
Foram encontrados erros enquanto processava:
 nvidia-dkms-470
 nvidia-driver-470
E: Sub-process /usr/bin/dpkg returned an error code (1)
`

Basically, errors state that "There was an error while processing nvidia-dkms-470". 
And after rebooting, everything stays pretty much the same.

leviport, 2021-12-14:
Hmm, if it's failing while Kernelstub is doing its thing, I have a hunch it might be related to bootloader troubles. It might be worth having a look at this article: https://support.system76.com/articles/bootloader/

jtmackoy, 2022-01-21:
I know this likely won't help @binaryz3r0, but after finding this issue when I encountered the same problem (could not switch from Integrated graphics), I thought I'd leave some thoughts for others or my future self when I rediscover this issue:

@leviport's recommendation:
> ```
> sudo apt purge ~nnvidia
> sudo apt clean && sudo apt autoremove
> sudo apt install nvidia-driver-470
> ```
worked for me, except, instead of expressly installing the `nvidia-driver-470` package, I opted for the `system76-driver-nvidia` package.  At this moment, I believe the two to be equivalent as the `system76-driver-nvidia` package will:

> [... pull in the latest NVIDIA drivers as packaged by System76, and all related packages needed ...](https://support.system76.com/articles/system76-driver/#installing-the-system76-nvidia-driver-for-systems-with-nvidia-gpus)

...which is great, right now.  That said, hopefully with this modification future me won't inadvertently install v470 of the driver when the latest is v560 or something like it (substituting `nvidia-driver-470` with `system76-driver-nvidia` *should* future proof it).

Also, since I'm running on system76 hardware this change seemed appropriate.

I struggled for awhile with `sudo apt purge ~nnvidia` not executing properly in my shell.  Then I realized, @leviport likely runs a bash shell (I run zsh).  A quick `exec bash` before running the commands he referenced (minus the last line I modified), then reboot, then switch to Hybrid graphics seems to have worked.

Thanks @leviport!

jtmackoy, 2022-02-21:
I've referenced this with others and thought it best to include an _explicit_ listing of what I run when this happens.  Here it is:

```bash
exec bash # This is used if you don't use bash for your shell
sudo apt purge ~nnvidia # Remove all nvidia-relevant packages
sudo apt clean # Remove apt cache
sudo apt update -m # Update package metadata, ignore missing packages
sudo dpkg --configure -a # Configure any packages not yet configured
sudo apt install -f # Fix any broken packages
sudo apt full-upgrade # Upgrade any packages that need it
sudo apt autoremove # Remove any unneeded packages still installed
sudo apt install system76-driver-nvidia # Install the nvidia driver via system76 package
sudo shutdown -r now # Reboot
# After reboot, check to see if mode doesn't automatically switch.
# If not, switch it and reboot again when prompted.
```

BTW - this still occurs for me.  Happened today.  I've opened a ticket with system76 support and they recommended a slightly different set of commands than [what I ran before](https://github.com/pop-os/system76-power/issues/296#issuecomment-1018184961).  Updated above with their recommendations.

disbr007, 2022-05-05:
This also happened to me today. @jtmackoy thanks very much for your notes - your first comment fixed it for me, though I'll go for your follow up/system76's recommendation next time.

NikhilJoshua, 2022-05-06:
Even I faced this issue today. The [first suggestion](https://github.com/pop-os/system76-power/issues/296#issuecomment-988083997) helped me.

HansN87, 2022-07-18:
experienced this issue today too. Unfortunately, @jtmackoy 's commands (https://github.com/pop-os/system76-power/issues/296#issuecomment-1047208380) didn't resolve it for me. I keep getting stuck at 
```
Errors were encountered while processing:
 nvidia-dkms-515
 nvidia-driver-515
 system76-driver-nvidia
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

cstrahan, 2022-11-29:
This worked for me: https://github.com/pop-os/system76-power/issues/296#issuecomment-1047208380

I'll not that I've been on the latest Pop_OS release since the week it came out. Haven't performed any updates in the past couple weeks. Just went into my office to find the fans spinning like crazy, and the system frozen. Rebooted and found that the system had switched to integrated graphics, and all attempts to switch to nvidia/discrete graphics failed. Had to perform the aforementioned steps to fix things.

Would be really great if the root cause of this could be figured out and resolved once and for all.

MihauKun, 2022-12-25:
> My machine occasionally does this when either the Nvidia driver or the kernel is updated. I'm usually able to fix it by purging and reinstalling the Nvidia driver like so:
> 
> ```
> sudo apt purge ~nnvidia
> sudo apt clean && sudo apt autoremove
> sudo apt install nvidia-driver-470
> ```

Thank You
That's worked for me

senntore, 2023-04-08:
I had this issue & it's fixed by following the steps mentioned above.
Thanks a lot!
It would be nice if there was a message or something to redirect to these steps if switching doesn't work.

AaronHo-0716, 2023-04-13:
[This](https://github.com/pop-os/system76-power/issues/296#issuecomment-1018184961) worked like a charm! Thanks a lot

nhodang99, 2023-08-06:
> I know this likely won't help @binaryz3r0, but after finding this issue when I encountered the same problem (could not switch from Integrated graphics), I thought I'd leave some thoughts for others [...]

Does not work for me :( I try many times but sudo system76-power graphics always give the same output "integrated".
One more ridiculous thing is that my laptop's battery is broken, so i need plug the charge to use laptop. In status bar, there is no battery detected (this is correct) but graphical UI to switch graphic mode doesn't show up too, this is strange, they should not be related.

tartley, 2024-11-21:
This procedure didn't work for me. I'm on latest stable popOS, went away from home for a week, came back to integrated graphics mode and can't switch away. The commands above (I did the "system76-driver-nvidia" variant) seemed to work, but on reboot I now find I can't login at the desktop. Sigh. Personally a separate problem unique to me...