pop-os/system76-power #299: Need to start `system76-power` manually on Fedora 35 (Workstation Edition)

nickgerace, 2021-12-12:
**Distribution (run `cat /etc/os-release`):**
```
Fedora Workstation 35 (Workstation Edition)
5.15.6-200.fc35.x86_64
```

**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
```
% sudo dnf info system76-power
Last metadata expiration check: 0:04:07 ago on Sun 12 Dec 2021 11:05:55 AM EST.
Installed Packages
Name         : system76-power
Version      : 0.50
Release      : 1.fc35
Architecture : x86_64
Size         : 3.6 M
Source       : system76-power-0.50-1.fc35.src.rpm
Repository   : @System
From repo    : copr:copr.fedorainfracloud.org:szydell:system76
Summary      : system76-power
URL          : https://github.com/pop-os/system76-power
License      : GPLv3
Description  : System76 Power Management daemon

Available Packages
Name         : system76-power
Version      : 0.50
Release      : 1.fc35
Architecture : src
Size         : 73 k
Source       : None
Repository   : copr:copr.fedorainfracloud.org:szydell:system76
Summary      : system76-power
URL          : https://github.com/pop-os/system76-power
License      : GPLv3
Description  : System76 Power Management daemon
```

**Issue/Bug Description:** When I start up my Thelio Major (Threadripper 3970X) and use Fedora normally (terminal, browsing, whatever), the fans consistently spin. I notice that if I run `systemctl start system76-power`, then the fans stop spinning unless the system is under load, which is the behavior I expect. The problem is that the service _is_ enabled.

_Hypothesis_: I suspect that there's a failure occurring upon system _startup_, and that once the distro is up and running and I am logged in, that issue is not present and I can successfully start the service.

_Relevant info:_
```
% systemctl list-unit-files | grep enabled | grep system76
system76-firmware-daemon.service              enabled         disabled
system76-power-wake.service                   enabled         disabled
system76-power.service                        enabled         disabled
```

_Anything "weird" in my environment?:_ My default shell is `zsh`, which is not the distro default. I could also see a scenario where the default shell is affecting this as well, but I suspect that is not the case..

I installed `zsh` using the following:
```bash
sudo dnf install -y zsh
chsh -s $(which zsh)
```

Another thing, I installed the [System76 Driver](https://support.system76.com/articles/system76-driver/) and all [System76 Software](https://support.system76.com/articles/system76-software/), including ACPI DKMS over DKMS, _except_ for the following:

1. System76 Power GNOME Shell Extension package -- I like to keep GNOME as vanilla as possible, so I _believe_ this is optional, but I could be wrong
2. System76 OLED package -- on a Thelio, so I do not need it

_Logs:_ I tried to parse the logs via `journalctl -u system76-power`, but they do not seem to line up with my experience. Here are the last two boots:

```
-- Boot Foo --
Dec 12 00:32:28 thelio systemd[1]: Starting System76 Power Daemon...
Dec 12 00:32:28 thelio system76-power[5584]: [INFO] Starting daemon
Dec 12 00:32:28 thelio system76-power[5584]: [INFO] Connecting to dbus system bus
Dec 12 00:32:28 thelio system76-power[5584]: [INFO] Rescanning PCI bus
Dec 12 00:32:28 thelio system76-power[5584]: [INFO] 0000:23:00.0: AMD graphics
Dec 12 00:32:28 thelio system76-power[5584]: [INFO] 0000:23:00.1: Function for 0000:23:00.0
Dec 12 00:32:28 thelio system76-power[5584]: [INFO] 0000:23:00.0: Function for 0000:23:00.0
Dec 12 00:32:28 thelio system76-power[5584]: [INFO] Disabling NMI Watchdog (for kernel debugg>
Dec 12 00:32:28 thelio system76-power[5584]: [INFO] Setting automatic graphics power
Dec 12 00:32:28 thelio system76-power[5584]: [WARN] Failed to set automatic graphics power: d>
Dec 12 00:32:28 thelio system76-power[5584]: [INFO] Initializing with the balanced profile
Dec 12 00:32:28 thelio system76-power[5584]: [INFO] Registering dbus name com.system76.PowerD>
Dec 12 00:32:28 thelio systemd[1]: Started System76 Power Daemon.
Dec 12 00:32:28 thelio system76-power[5584]: [INFO] Adding dbus path /com/system76/PowerDaemo>
Dec 12 00:32:28 thelio system76-power[5584]: [INFO] Handling dbus requests
Dec 12 00:32:28 thelio system76-power[5584]: [ERROR] hid_backlight: no system76_acpi::kbd_bac>
Dec 12 00:44:37 thelio system76-power[5584]: [INFO] caught signal: SIGTERM
Dec 12 00:44:37 thelio systemd[1]: Stopping System76 Power Daemon...
Dec 12 00:44:38 thelio system76-power[5584]: [INFO] daemon exited from loop
Dec 12 00:44:38 thelio systemd[1]: system76-power.service: Deactivated successfully.
Dec 12 00:44:38 thelio systemd[1]: Stopped System76 Power Daemon.

-- Boot Bar --
Dec 12 11:05:48 thelio systemd[1]: Starting System76 Power Daemon...
Dec 12 11:05:48 thelio systemd[1]: system76-power.service: Deactivated successfully.
Dec 12 11:05:48 thelio systemd[1]: Stopped System76 Power Daemon.
Dec 12 11:05:48 thelio systemd[1]: Starting System76 Power Daemon...
Dec 12 11:05:48 thelio system76-power[6736]: [INFO] Starting daemon
Dec 12 11:05:48 thelio system76-power[6736]: [INFO] Connecting to dbus system bus
Dec 12 11:05:48 thelio system76-power[6736]: [INFO] Rescanning PCI bus
Dec 12 11:05:48 thelio system76-power[6736]: [INFO] 0000:23:00.0: AMD graphics
Dec 12 11:05:48 thelio system76-power[6736]: [INFO] 0000:23:00.1: Function for 0000:23:00.0
Dec 12 11:05:48 thelio system76-power[6736]: [INFO] 0000:23:00.0: Function for 0000:23:00.0
Dec 12 11:05:48 thelio system76-power[6736]: [INFO] Disabling NMI Watchdog (for kernel debugg>
Dec 12 11:05:48 thelio system76-power[6736]: [INFO] Setting automatic graphics power
Dec 12 11:05:48 thelio system76-power[6736]: [WARN] Failed to set automatic graphics power: d>
Dec 12 11:05:48 thelio system76-power[6736]: [INFO] Initializing with the balanced profile
Dec 12 11:05:48 thelio system76-power[6736]: [INFO] Registering dbus name com.system76.PowerD>
Dec 12 11:05:48 thelio systemd[1]: Started System76 Power Daemon.
Dec 12 11:05:48 thelio system76-power[6736]: [INFO] Adding dbus path /com/system76/PowerDaemo>
Dec 12 11:05:48 thelio system76-power[6736]: [INFO] Handling dbus requests
Dec 12 11:05:48 thelio system76-power[6736]: [ERROR] hid_backlight: no system76_acpi::kbd_bac
```

**Steps to reproduce (if you know):**
1. Install Fedora 35 (Workstation Edition, the standard/vanilla edition) on System76 hardware (I presume, Coreboot-enabled hardware)
2. Install the `system76-driver` and System76 Software packages on Fedora except for the Power GNOME Shell extension and OLED screen packages (and ACPI DKMS instead of DKMS)
3. Execute `systemctl enable system76-power` (or with the `--now` flag, or followed by `systemctl start system76-power)
4. Restart the system and observe moderately high and consistent fan noise (independent of task)
5. Once logged in, execute `systemctl start system76-power` and notice the system become immediately silent

**Expected behavior:** The `system76-power` package works upon startup on Fedora 35 and the Thelio Major fans do not spin up.

**Other Notes:**
Related issue: https://github.com/pop-os/system76-power/issues/268

I installed the `system76-power` package from the official docs.
I am actively making changes to the docs if there's something else I can add to help out!

nickgerace, 2021-12-12:
Just reproduced again to double check and I can confirm the reproduction steps work for my machine. However, I have now narrowed down the logs to help.

This is what I see once I am logged in and going about my day:
```
% journalctl -u system76-power -b
-- Journal begins at Fri 2021-12-10 18:29:38 EST, ends at Sun 2021-12-12 11:41:41 EST. --
-- No entries --
```

Now, let's start the service.
```
% sudo systemctl start system76-power
```

I immediately hear the fans go silent. Seems to work fine!

```
% journalctl -u system76-power -b
-- Journal begins at Fri 2021-12-10 18:29:38 EST, ends at Sun 2021-12-12 11:42:17 EST. --
Dec 12 11:42:17 thelio systemd[1]: Starting System76 Power Daemon...
Dec 12 11:42:17 thelio system76-power[5044]: [INFO] Starting daemon
Dec 12 11:42:17 thelio system76-power[5044]: [INFO] Connecting to dbus system bus
Dec 12 11:42:17 thelio system76-power[5044]: [INFO] Rescanning PCI bus
Dec 12 11:42:17 thelio system76-power[5044]: [INFO] 0000:23:00.0: AMD graphics
Dec 12 11:42:17 thelio system76-power[5044]: [INFO] 0000:23:00.1: Function for 0000:23:00.0
Dec 12 11:42:17 thelio system76-power[5044]: [INFO] 0000:23:00.0: Function for 0000:23:00.0
Dec 12 11:42:17 thelio system76-power[5044]: [INFO] Disabling NMI Watchdog (for kernel debugging only)
Dec 12 11:42:17 thelio system76-power[5044]: [INFO] Setting automatic graphics power
Dec 12 11:42:17 thelio system76-power[5044]: [WARN] Failed to set automatic graphics power: does not have switchable graphics
Dec 12 11:42:17 thelio system76-power[5044]: [INFO] Initializing with the balanced profile
Dec 12 11:42:17 thelio system76-power[5044]: [INFO] Registering dbus name com.system76.PowerDaemon
Dec 12 11:42:17 thelio system76-power[5044]: [INFO] Adding dbus path /com/system76/PowerDaemon with interface com.system76.PowerDaemon
Dec 12 11:42:17 thelio system76-power[5044]: [INFO] Handling dbus requests
Dec 12 11:42:17 thelio systemd[1]: Started System76 Power Daemon.
Dec 12 11:42:17 thelio system76-power[5044]: [ERROR] hid_backlight: no system76_acpi::kbd_backlight led
```

Things generally look good at this point. Still investigating, but I think this issue _should_ be pretty reproducible at this point.

linuxpng, 2021-12-13:
I think I understand what you're saying and this is because system76-power conflicts with power-profiles-daemon.  It's not enough to disable the power-profiles-daemon, you need to uninstall it then blacklist it from /etc/dnf/dnf.conf

excludepkgs=power-profiles-daemon

I'm not wild about being dependent on a COPR, that may go away at anytime, for fan control.  It may be worth moving the fans from the i/o board to the system board.

nickgerace, 2021-12-13:
> I think I understand what you're saying and this is because system76-power conflicts with power-profiles-daemon. It's not enough to disable the power-profiles-daemon, you need to uninstall it then blacklist it from /etc/dnf/dnf.conf
> 
> excludepkgs=power-profiles-daemon

Ah I see. Perhaps, this would be good to add to the docs then?
It should likely have instructions to reverse it as well.

> I'm not wild about being dependent on a COPR, that may go away at anytime, for fan control. It may be worth moving the fans from the i/o board to the system board.

That would be a little sad if the COPR went away since I prefer Fedora, but I understand.
What would be the alternative for Fedora users who already own a System76 machine?

linuxpng, 2021-12-13:
I don't know.  Right know we're dependent on szydell maintaining interest in Fedora.  I did ask system76 if they had plans to introduce fan control into power-profiles-daemon and they said 'no'.  I'm not even sure my question makes sense with what power-profiles-daemon does.  If the COPR goes away we'd be on the hook for building it ourselves.  This project could probably benefit from instructions on how to do that, but maybe I just haven't found them yet.

ahoneybun, 2021-12-13:
It looks like this may be with GNOME 41 as I see the same behavior on Arch:

https://github.com/pop-os/system76-power/issues/294

linuxpng, 2021-12-13:
Hey Aaron, yeah, gnome 41 switched to power-profiles-daemon by default so Arch users would have to do the same thing.  Is there any chance system76 would make the default fan behavior on the thelio i/o board firmware to use the fan curve instead of the 100% on the connected fans?  Also, are there any plans to get the thelio i/o driver in the mainline kernel?  These are both from the wish to be able to have any linux distro on the thelios w/o being dependent upon compiling the drivers ourselves or community good will to do so for the hardware.

ahoneybun, 2021-12-13:
1. The fan curves are set by the system76-power and I don't think that will be changing.

2. I don't know at this time.

ahoneybun, 2021-12-13:
> I think I understand what you're saying and this is because system76-power conflicts with power-profiles-daemon.  It's not enough to disable the power-profiles-daemon, you need to uninstall it then [...]

I won't recommend moving to the system board as you lose the benefits of the Io board. You also will not benefit from updates when it comes to fan curves.

linuxpng, 2021-12-13:
> 1. The fan curves are set by the system76-power and I don't think that will be changing.
> 
>     2. I don't know at this time.

That's unfortunate, hopefully System76 reconsiders this.  We obviously cannot expect you to maintain versions of the drivers for other distributions, but non developer-skilled users may end up losing out on basic functionality if they do not want to use Pop OS.

ahoneybun, 2021-12-13:
Which one are you commenting on? Just because I'm not aware of working with mainline to add a feature doesn't mean that it's not being worked on or talked about. You can also use this software on Ubuntu or any other Ubuntu based OS as well not just Pop!_OS.

linuxpng, 2021-12-13:
> Which one are you commenting on? Just because I'm not aware of working with mainline to add a feature doesn't mean that it's not being worked on or talked about. You can also use this software on Ubuntu or any other Ubuntu based OS as well not just Pop!_OS.

Both system76-power and the thelio-io driver.  Both are required for proper fan control, correct?  I am really not trying to be difficult, but my workflow depends on Fedora so I know as long as the COPR is there I am fine.  There's no other motive than to make sure that Fedora's rapid upgrade cycle doesn't burn me.

ahoneybun, 2021-12-13:
I don't believe you are being difficult at all I just wanted to make sure I understood you.

ahoneybun, 2021-12-13:
The data that you looked at @nickgerace and that I looked at are the same locations. I also believed it was the new power-profiles that GNOME 41 added though I didn't remove the package and only disabled the service which didn't fix it.

HennadiiM, 2021-12-13:
> I am starting to believe this might be related to GNOME 41 given the issue I linked and the one you did as well @ahoneybun. Do we know if there's a logging location to look at so we can see what's going on?

Btw, I'm having the same issue, but with KDE on distro Garuda (arch-based)

nickgerace, 2021-12-13:
Ahhhh that would make sense. An eventual fix would be required for an eventual Pop_OS! 22.04 LTS since I'd assume it will also have GNOME 41+ by that point.

tunix, 2022-01-18:
On Fedora Silverblue 35 (running on lemp10), the daemon crashes:

```
-- Boot f01c29eb29ec4ee5892f8a18c52df292 --
Oca 18 12:20:46 fedora systemd[1]: Starting System76 Power Daemon...
Oca 18 12:20:46 fedora system76-power[19165]: [INFO] Starting daemon
Oca 18 12:20:46 fedora system76-power[19165]: [INFO] Connecting to dbus system bus
Oca 18 12:20:46 fedora system76-power[19165]: [INFO] Rescanning PCI bus
Oca 18 12:20:46 fedora system76-power[19165]: [INFO] 0000:00:02.0: Intel graphics
Oca 18 12:20:46 fedora system76-power[19165]: [INFO] 0000:00:02.0: Function for 0000:00:02.0
Oca 18 12:20:46 fedora system76-power[19165]: [INFO] Disabling NMI Watchdog (for kernel debugging only)
Oca 18 12:20:46 fedora system76-power[19165]: [INFO] Setting automatic graphics power
Oca 18 12:20:46 fedora system76-power[19165]: [WARN] Failed to set automatic graphics power: does not have switchable graphics
Oca 18 12:20:46 fedora system76-power[19165]: [INFO] Initializing with the balanced profile
Oca 18 12:20:46 fedora system76-power[19165]: [INFO] Registering dbus name com.system76.PowerDaemon
Oca 18 12:20:46 fedora system76-power[19165]: [INFO] Adding dbus path /com/system76/PowerDaemon with interface com.system76.PowerDaemon
Oca 18 12:20:46 fedora systemd[1]: Started System76 Power Daemon.
Oca 18 12:20:46 fedora system76-power[19165]: [ERROR] fan daemon: platform hwmon not found
Oca 18 12:20:46 fedora system76-power[19165]: [INFO] Handling dbus requests
Oca 18 12:20:46 fedora system76-power[19165]: thread '<unnamed>' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 2, kind: NotFound, message: "No such file or directory" }', src/hid_backlight.rs:77:55
Oca 18 12:20:46 fedora system76-power[19165]: note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Oca 18 13:18:01 fedora systemd[1]: Stopping System76 Power Daemon...
Oca 18 13:18:01 fedora system76-power[19165]: [INFO] caught signal: SIGTERM
Oca 18 13:18:02 fedora system76-power[19165]: [INFO] daemon exited from loop
Oca 18 13:18:02 fedora systemd[1]: system76-power.service: Deactivated successfully.
Oca 18 13:18:02 fedora systemd[1]: Stopped System76 Power Daemon.
Oca 18 13:18:02 fedora systemd[1]: Starting System76 Power Daemon...
```

I can't see fan related information in `Vitals` extension nor in the output of `sensors` command. I wasn't able to install DKMS modules of system76 due to Silverblue being an ostree based distro. Could these be related?

I haven't done a battery test yet so not really sure how different this configuration will be compared to Pop!_OS.

fnichol, 2022-02-08:
I'm running Arch Linux with GNOME 41 and I may have figured out the interaction that's triggering the behavior of `system76-power` shutting down.

It looks like GNOME 41 ships with power profiles support via [power-profiles-daemon](https://gitlab.freedesktop.org/hadess/power-profiles-daemon/-/blob/main/README.md) which has a [Why Not/system76-power](https://gitlab.freedesktop.org/hadess/power-profiles-daemon/-/blob/main/README.md#system76-power) paragraph explaining that these 2 projects are "very similar" so a potential interaction is known to the power-profiles-daemon developers at a minimum.

Additionally in the same README in the [Conflicts](https://gitlab.freedesktop.org/hadess/power-profiles-daemon/-/blob/main/README.md#conflicts) section they link directly to the `Conflicts=` line in the `power-profiles.daemon.service` systemd unit file.

I know that this service is not active, but however is loaded:

```console
> systemctl is-active power-profiles-daemon
inactive

> systemctl status power-profiles-daemon
○ power-profiles-daemon.service - Power Profiles daemon
     Loaded: loaded (/usr/lib/systemd/system/power-profiles-daemon.service; disabled; vendor preset: disabled)
     Active: inactive (dead)
```

When I start a GNOME session (i.e. login via LightDM or equivalent), then check the journald logs for `power-profiles.daemon.service` and `system76-power` I can see that `power-profiles-daemon` did run and that at about the same time, the `system76-power` service gets its `SIGTERM` and shuts down.

So, I decided to try and mask the `power-profiles-daemon` unit with `sudo systemctl status power-profiles-daemon`, then completely rebooted my system. So far...I think I'm good!

---

In preparing this writeup, I came across this issue on `power-profiles-daemon`'s issue tracker: [Fan support on a System76 Thelio](https://gitlab.freedesktop.org/hadess/power-profiles-daemon/-/issues/66). The submitter of the issue notes the same interaction with these 2 services and asks about a remedy, and one of the project maintainers suggests that `power-profiles-daemon` can't do much to remedy the situation and that a "split off" daemon would likely be the use case here (I presume they are referring to `system76-power` or equivalent).

So it might be that our solution for the moment is to mask `power-profiles-daemon.service` and carry on?

nickgerace, 2022-02-08:
Thanks @fnichol! Another discovery as well on Fedora 35 Workstation: if you change the suspend idle settings (e.g. "suspend after 1 hour idle"), then the power profiles daemon is restarted. Starting `system76-power` causes works again in that scenario.

Regardless, in your scenario and mine, it seems masking the power profiles daemon service is the move for the time being.

nekojet, 2022-02-10:
> So, I decided to try and mask the `power-profiles-daemon` unit with `sudo systemctl status power-profiles-daemon`, then completely rebooted my system. So far...I think I'm good!

I could be wrong, but I think you meant `sudo systemctl mask power-profiles-daemon` to mask the service.  In any case, thanks for hunting this down!  I've now masked `power-profiles-daemon` to get `system76-power` working (for now) on Fedora 35.  Cheers!

tunix, 2022-03-31:
Hi,

As we're approaching 22.04 LTS release and recent developments on system76-power, do we now have a solution? Are we now able to user system76-power on Fedora Workstation and Silverblue?

jppelt, 2022-04-06:
Thanks to all who solved this.  I masked and my Thelio R2 is as quiet as ever.  Not sure why it took me so long to get to Fedora but 35 is just so nice.  Smooth.  Now it is quiet.

ahoneybun, 2022-04-06:
> Hi,
> 
> As we're approaching 22.04 LTS release and recent developments on system76-power, do we now have a solution? Are we now able to use system76-power on Fedora Workstation and Silverblue?

I haven't seen a change on Arch yet but there is work on updating the packaging which may help:

https://github.com/pop-os/system76-power/pull/327

spcbfr, 2022-12-09:
Any updates on this? almost the exact same thing is happening on nixos https://discourse.nixos.org/t/very-high-fan-noises-on-nixos-using-a-system76-thelio/23875/6

ahoneybun, 2022-12-09:
> Any updates on this? almost the exact same thing is happening on nixos https://discourse.nixos.org/t/very-high-fan-noises-on-nixos-using-a-system76-thelio/23875/6

This is more about a conflict with GNOME's new power options and system76-power's own settings so it'll take time.

spcbfr, 2022-12-09:
@ahoneybun 
you are still getting the same error as me tho, so it's kinda related maybe?

```
Dec 12 11:05:48 thelio system76-power[6736]: [ERROR] hid_backlight: no system76_acpi::kbd_bac # we're both getting this error about keyboards
```

ahoneybun, 2022-12-09:
I don't have a Thelio so I can't confirm there.

rsoeldner, 2023-07-11:
@spcbfr any solution to this? I'm running into the same problem and find myself frequently running `systemctl restart system76-power.service`