pop-os/system76-power #300: Changing the power profile doesn't change anything

Tom743, 2021-12-15:
**Distribution:**
Pop!_OS 21.04
Kernel: 5.15.5-76051505-generic

**Hardware:**
Host: HP Laptop 15-dw3xxx
CPU: 11th Gen Intel i5-1135G7 (8) @ 4.200GHz 
GPU: Intel TigerLake GT2 [Iris Xe Graphics]

I remember the same thing happening on a 10th Gen Intel i5-1035G1 Lenovo Ideapad 3 laptop with Pop!_OS 20.04 LTS
 
**Related Application and/or Package Version:**
system76-power:
  Installed: 1.1.20\~1637635456\~21.04~b5ed171

**Issue/Bug Description:**
Changing the power mode won't alter the cpu frequency cap or the screen brightness. I realized that a few hours ago while reading about the expected behaviour (reduced screen brightness and cpu frequency for lower power modes) and I realized I had never seen any change in brightness variation or noticeable performance drop when using battery life mode since I started using `Pop!_OS` about a year ago (in 2 different laptops and 3 installs in total).

`system76-power profile` output for different power modes:
```
Power Profile: Performance
CPU: 9% - 100%, Turbo
Backlight intel_backlight: 33312/96000 = 34%

Power Profile: Balanced
CPU: 9% - 100%, Turbo
Backlight intel_backlight: 33312/96000 = 34%

Power Profile: Battery
CPU: 9% - 100%, Turbo
Backlight intel_backlight: 33312/96000 = 34%
```

Neither `powertop` or `tlp` are installed, so I don't see a reason for this to be the current configuration. I could't find any configuration file that I can tune to correct it.

**Steps to reproduce:**
Run `system76-power profile` in the three different profiles and compare the output

**Expected behavior:**
Lower maximum frequency, no turbo enabled and lower built in screen brightness on battery mode, and higher lower bound for CPU frequency on performance mode.

Tom743, 2021-12-15:
I just updated to Pop!_OS 20.10 and I still get the same behaviour

Tom743, 2021-12-16:
I found the file [src/daemon/profiles.rs](https://github.com/pop-os/system76-power/blob/master/src/daemon.rs) in this repository which seems to be responsible of the pstate settings. I have no reason to believe that the file might have been modified at some point of the instalation, but just to be sure, where is it supposed to be located on my drive? It probably won't be of any help if I find it but I'm just curious to know where these files are located.
I don't think the frequency controller module (if such thing exists) is broken because I just installed [this](https://extensions.gnome.org/extension/945/cpu-power-manager/) extension and works fine. I can disable turbo boost and change the minimum and maximum values for the CPU frequency and it does its job. I don't know if that helps, maybe these two extensions are using different drivers?

dxdns, 2023-07-14:
Sabe problem here

Acer Aspire A515-52G
Pop OS 22.04 LTS

dxdns, 2023-07-14:
Same problem here

Acer Aspire A515-52G
Pop OS 22.04 LTS