pop-os/system76-power #313: Nvidia mode not setting nvidia as main display

CygnusTerminal, 2022-03-02:
**Distribution (run `cat /etc/os-release`):**
 NAME="Arch Linux"
PRETTY_NAME="Arch Linux"
ID=arch
BUILD_ID=rolling
ANSI_COLOR="38;2;23;147;209"
HOME_URL="https://archlinux.org/"
DOCUMENTATION_URL="https://wiki.archlinux.org/"
SUPPORT_URL="https://bbs.archlinux.org/"
BUG_REPORT_URL="https://bugs.archlinux.org/"

 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**

**Issue/Bug Description:**
When switching to nvidia mode, main screen is still the AMD GPU-driven display. So, the nvidia-only mode it's non-functional. Also, when checking the Xorg directory there is no xorg file.

**Steps to reproduce (if you know):**
Switch to nvidia-mode
Reboot
Check with xrandr

**Expected behavior:**
When setting nvidia mode, it should create an Xorg file to set the nvidia gpu as the main GPU.

**Other Notes:**

crawfxrd, 2022-03-02:
Duplicate of #148
Requires #291

closed 2022-03-02 by crawfxrd

CygnusTerminal, 2022-03-03:
Thanks for the heads up!