pop-os/system76-power #320: Archlinux: system76-power is unable to set integrated mode

CygnusTerminal, 2022-03-19:
**Distribution (run `cat /etc/os-release`):**
NAME="Arch Linux"
PRETTY_NAME="Arch Linux"
ID=arch
BUILD_ID=rolling
ANSI_COLOR="38;2;23;147;209"
HOME_URL="https://archlinux.org/"
DOCUMENTATION_URL="https://wiki.archlinux.org/"
SUPPORT_URL="https://bbs.archlinux.org/"
BUG_REPORT_URL="https://bugs.archlinux.org/"
LOGO=archlinux-logo
 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
aur/system76-power

**Issue/Bug Description:**
When setting `integrated `mode and rebooting, instead of setting integrated as expected, and powering down the dGPU, it sets the mode to compute, with the dGPU still working and using power.

**Steps to reproduce (if you know):**
1. Set gpu mode to `integrated`
2. Reboot
3. Check gpu mode
4. Observe `compute` mode instean of the one desired

**Expected behavior:**
GPU should power down and start using the integrated GPU

**Other Notes:**
It does the same with the -git version.
Lenovo Ideapad Gaming 3 ARH015 with GTX 1650 GDDR6

mmstick, 2022-03-19:
Did you report the issue to the AUR maintainer?

CygnusTerminal, 2022-03-19:
> Did you report the issue to the AUR maintainer?

Hello, yes, awaiting a reply from the maintainer.

CygnusTerminal, 2022-03-19:
Sorry to spam. I have found that i am not the only one with this issue on the AUR page.

jacobgkau, 2022-03-21:
This is a duplicate of https://github.com/pop-os/system76-power/issues/252.

closed 2022-03-21 by jacobgkau