pop-os/system76-power #322: Compilation fails for 32-bit

ErichRitz, 2022-03-23:
**Distribution (run `cat /etc/os-release`):**
```
NAME=Slackware
VERSION="15.0"
ID=slackware
VERSION_ID=15.0
PRETTY_NAME="Slackware 15.0 i586"
ANSI_COLOR="0;34"
CPE_NAME="cpe:/o:slackware:slackware_linux:15.0"
HOME_URL="http://slackware.com/"
SUPPORT_URL="http://www.linuxquestions.org/questions/slackware-14/"
BUG_REPORT_URL="http://www.linuxquestions.org/questions/slackware-14/"
VERSION_CODENAME=stable
```
 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**

system76-power-1.1.20

**Issue/Bug Description:**

Compiling system76-power fails with:

```
   Compiling system76-power v1.1.20 (/tmp/SBo/system76-power-1.1.20)
error[E0308]: mismatched types
  --> src/sideband.rs:45:13
   |
45 |             sbreg_phys as i64,
   |             ^^^^^^^^^^^^^^^^^ expected `i32`, found `i64`
   |
help: you can convert an `i64` to an `i32` and panic if the converted value doesn't fit
   |
45 |             (sbreg_phys as i64).try_into().unwrap(),
   |             +                 +++++++++++++++++++++

For more information about this error, try `rustc --explain E0308`.
error: could not compile `system76-power` due to previous error
warning: build failed, waiting for other jobs to finish...
error: build failed
make: *** [Makefile:64: target/release/system76-power] Error 101
```

**Steps to reproduce (if you know):**

On 32-bit Slackware, try to run the SlackBuild here:
https://slackbuilds.org/repository/15.0/system/system76-power/

**Expected behavior:**

Compile successfully.

**Other Notes:**

I have not tried to compile on 32-bit Pop_OS!, but I suspect the results are the same.

crawfxrd, 2022-03-23:
> I have not tried to compile on 32-bit Pop_OS

Pop!_OS only supports x86_64 and AArch64.

ErichRitz, 2022-03-23:
Actually it looks to be more complicated.  Earlier in sideband.rs we have
```
pub const PCR_BASE_ADDRESS: usize = 0xFD00_0000;
```
So I'm just going to mark 32-bit as unsupported.

closed 2022-03-29 by ErichRitz