pop-os/system76-power #325: GPU fan runs nonstop after update to version 1.1.20~1648241921~21.10~1e1599a

n3m0-22, 2022-03-30:
**System:**
NAME="Pop!_OS"
VERSION="21.10"
```oryp4``` 

``` 
**Software:** 
Installed: 1.1.20~1648241921~21.10~1e1599a
Candidate: 1.1.20~1648241921~21.10~1e1599a
Version table:
*** 1.1.20~1648241921~21.10~1e1599a 1001
     1001 http://apt.pop-os.org/release impish/main amd64 Packages
      100 /var/lib/dpkg/status
```

**Bug:**
Upgrading ```system76-power``` on the orxy pro 4 from version ```1.1.20~1647360548~21.10~815500d``` to the current release causes the GPU fan to run constantly no matter the temperature or CPU load. This happens under all GPU settings.

**Expected behavior:**
Normally with graphics set to integrated the GPU fan only turns on for a few seconds on occasion as pictured below

[image]

**Actual behavior:**
After installing version ```1.1.20~1648241921~21.10~1e1599a```
[image]

**Other Notes:**
Using ```Fn + 1``` will only reset it for a short time before it starts up again

[image]

**Steps to Replicate:**
*Note so far this only happens on the ```oryp4```*

1. Fresh install of ```pop-os_21.10_amd64_nvidia_7```
2. Set Graphics mode to Integrated and restart then run the following
3. ```sudo apt-mark hold system76-power```
4. ```sudo apt update```
5. ```sudo apt upgrade```
6. Use psensor or lm-sensors to inspect GPU fan speeds
7. ```sudo apt-mark unhold system76-power```
8. ```sudo apt upgrade```
9. Now with the newest version of system76-power installed compare the fan speed

crawfxrd, 2022-03-30:
Does reverting 1e1599a fix it?

n3m0-22, 2022-03-30:
Yup that fixed it! Thank you.

crawfxrd, 2022-03-30:
Can you verify `/sys/bus/pci/devices/0000:01:00.0/power/control` is set to `auto` in integrated mode?

n3m0-22, 2022-03-30:
Both return No such file or directory

n3m0-22, 2022-03-30:
The dir ```0000:01:00.0``` is not there at al
[image]
l

crawfxrd, 2022-03-30:
If you update the package again to the released version it should be there in integrated mode.

The jq command will need you to change the directory for the NVIDIA driver files So if you have 510 installed it would be `nvidia-driver-510` instead of `nvidia-driver-470`.

n3m0-22, 2022-03-30:
After updating the package to the released version
```/sys/bus/pci/devices/0000:01:00.0/power/control``` does return ```auto```

The jq command returns nothing no matter the graphics mode I set. I checked and the supported-gpus.json file is there but there is no devid entry in it for 0000:01:00.0.

crawfxrd, 2022-03-30:
The ID you need is whatever is *in* `/sys/bus/pci/devices/0000:01:00.0/device`, not the PCI mapping itself.

You should be able to copy+paste the command, and maybe only have to change the driver version (e.g., `nvidia-driver-470` to `nvidia-driver-510`).

n3m0-22, 2022-03-30:
I never could get an output with the ```nvidia-driver-470``` so I updated to the ```nvidia-driver-510``` here is the output

```
{
  "devid": "0x1BA1",
  "name": "NVIDIA GeForce GTX 1070",
  "features": [
    "dpycbcr420",
    "dpgsynccompatible",
    "hdmi4k60rgb444",
    "geforce",
    "vdpaufeaturesetH2"
  ]
}
{
  "devid": "0x1BA1",
  "subdevid": "0x08A1",
  "subvendorid": "0x1028",
  "name": "NVIDIA GeForce GTX 1070 with Max-Q Design",
  "features": [
    "dpycbcr420",
    "dpgsynccompatible",
    "hdmi4k60rgb444",
    "geforce",
    "vdpaufeaturesetH2"
  ]
}
{
  "devid": "0x1BA1",
  "subdevid": "0x08A2",
  "subvendorid": "0x1028",
  "name": "NVIDIA GeForce GTX 1070 with Max-Q Design",
  "features": [
    "dpycbcr420",
    "dpgsynccompatible",
    "hdmi4k60rgb444",
    "geforce",
    "vdpaufeaturesetH2"
  ]
}
{
  "devid": "0x1BA1",
  "subdevid": "0x1CCE",
  "subvendorid": "0x1043",
  "name": "NVIDIA GeForce GTX 1070 with Max-Q Design",
  "features": [
    "dpycbcr420",
    "dpgsynccompatible",
    "hdmi4k60rgb444",
    "geforce",
    "vdpaufeaturesetH2"
  ]
}
{
  "devid": "0x1BA1",
  "subdevid": "0x1651",
  "subvendorid": "0x1458",
  "name": "NVIDIA GeForce GTX 1070 with Max-Q Design",
  "features": [
    "dpycbcr420",
    "dpgsynccompatible",
    "hdmi4k60rgb444",
    "geforce",
    "vdpaufeaturesetH2"
  ]
}
{
  "devid": "0x1BA1",
  "subdevid": "0x1653",
  "subvendorid": "0x1458",
  "name": "NVIDIA GeForce GTX 1070 with Max-Q Design",
  "features": [
    "dpycbcr420",
    "dpgsynccompatible",
    "hdmi4k60rgb444",
    "geforce",
    "vdpaufeaturesetH2"
  ]
}
{
  "devid": "0x1BA1",
  "subdevid": "0x11E8",
  "subvendorid": "0x1462",
  "name": "NVIDIA GeForce GTX 1070 with Max-Q Design",
  "features": [
    "dpycbcr420",
    "dpgsynccompatible",
    "hdmi4k60rgb444",
    "geforce",
    "vdpaufeaturesetH2"
  ]
}
{
  "devid": "0x1BA1",
  "subdevid": "0x11E9",
  "subvendorid": "0x1462",
  "name": "NVIDIA GeForce GTX 1070 with Max-Q Design",
  "features": [
    "dpycbcr420",
    "dpgsynccompatible",
    "hdmi4k60rgb444",
    "geforce",
    "vdpaufeaturesetH2"
  ]
}
{
  "devid": "0x1BA1",
  "subdevid": "0x1225",
  "subvendorid": "0x1462",
  "name": "NVIDIA GeForce GTX 1070 with Max-Q Design",
  "features": [
    "dpycbcr420",
    "dpgsynccompatible",
    "hdmi4k60rgb444",
    "geforce",
    "vdpaufeaturesetH2"
  ]
}
{
  "devid": "0x1BA1",
  "subdevid": "0x1226",
  "subvendorid": "0x1462",
  "name": "NVIDIA GeForce GTX 1070 with Max-Q Design",
  "features": [
    "dpycbcr420",
    "dpgsynccompatible",
    "hdmi4k60rgb444",
    "geforce",
    "vdpaufeaturesetH2"
  ]
}
{
  "devid": "0x1BA1",
  "subdevid": "0x1227",
  "subvendorid": "0x1462",
  "name": "NVIDIA GeForce GTX 1070 with Max-Q Design",
  "features": [
    "dpycbcr420",
    "dpgsynccompatible",
    "hdmi4k60rgb444",
    "geforce",
    "vdpaufeaturesetH2"
  ]
}
{
  "devid": "0x1BA1",
  "subdevid": "0x9501",
  "subvendorid": "0x1558",
  "name": "NVIDIA GeForce GTX 1070 with Max-Q Design",
  "features": [
    "dpycbcr420",
    "dpgsynccompatible",
    "hdmi4k60rgb444",
    "geforce",
    "vdpaufeaturesetH2"
  ]
}
{
  "devid": "0x1BA1",
  "subdevid": "0x95E1",
  "subvendorid": "0x1558",
  "name": "NVIDIA GeForce GTX 1070 with Max-Q Design",
  "features": [
    "dpycbcr420",
    "dpgsynccompatible",
    "hdmi4k60rgb444",
    "geforce",
    "vdpaufeaturesetH2"
  ]
}
{
  "devid": "0x1BA1",
  "subdevid": "0x2000",
  "subvendorid": "0x1A58",
  "name": "NVIDIA GeForce GTX 1070 with Max-Q Design",
  "features": [
    "dpycbcr420",
    "dpgsynccompatible",
    "hdmi4k60rgb444",
    "geforce",
    "vdpaufeaturesetH2"
  ]
}
{
  "devid": "0x1BA1",
  "subdevid": "0x1032",
  "subvendorid": "0x1D05",
  "name": "NVIDIA GeForce GTX 1070 with Max-Q Design",
  "features": [
    "dpycbcr420",
    "dpgsynccompatible",
    "hdmi4k60rgb444",
    "geforce",
    "vdpaufeaturesetH2"
  ]
}
```

PR #326 (2022-05-03): Disable dGPU in integrated mode if it doesn't support runtimepm
If a device does not support runtime power management, then remove it from the bus when in integrated mode.

Fixes: 1e1599a9dd12 ("daemon: Always enable GPU power")
Resolves: #325