pop-os/system76-power #332: system76-power broken after update/upgrade on 22.04

n3m0-22, 2022-04-17:
#### Steps to reproduce

- Clean install of ```pop-os_22.04_amd64_nvidia_2.iso```
- ```sudo apt update && sudo apt upgrade```
- ```sudo reboot```

Error in Extensions

```text
Gio.DBusError: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name com.system76.PowerDaemon was not provided by any .service files
```

Error in terminal

```bash
system76-power graphics
daemon returned an error message: "The name com.system76.PowerDaemon was not provided by any .service files"
```

After running ```sudo systemctl restart system76-power.service``` terminal output of ```system76-power graphics``` works correctly, but after restarting GNOME it breaks again.

```bash
systemctl status system76-power
○ system76-power.service - System76 Power Daemon
     Loaded: loaded (/lib/systemd/system/system76-power.service; enabled; vendo>
     Active: inactive (dead) since Sun 2022-04-17 10:07:36 MST; 12min ago
    Process: 6346 ExecStart=/usr/bin/system76-power daemon (code=exited, status>
   Main PID: 6346 (code=exited, status=0/SUCCESS)
        CPU: 35ms

Apr 17 10:07:15 pop-os system76-power[6346]: [ERROR] fan daemon: platform hwmon>
Apr 17 10:07:15 pop-os system76-power[6346]: [INFO] Handling dbus requests
Apr 17 10:07:15 pop-os system76-power[6346]: [INFO] hid_backlight: no devices f>
Apr 17 10:07:18 pop-os system76-power[6346]: [INFO] DBUS Received GetGraphics()>
Apr 17 10:07:20 pop-os system76-power[6346]: [INFO] Setting power management to>
Apr 17 10:07:36 pop-os systemd[1]: Stopping System76 Power Daemon...
Apr 17 10:07:36 pop-os system76-power[6346]: [INFO] caught signal: SIGTERM
Apr 17 10:07:36 pop-os system76-power[6346]: [INFO] daemon exited from loop
Apr 17 10:07:36 pop-os systemd[1]: system76-power.service: Deactivated successf>
Apr 17 10:07:36 pop-os systemd[1]: Stopped System76 Power Daemon.
```

#### Before update/upgrade

https://user-images.githubusercontent.com/11048214/163727133-11a9f200-7f13-4077-91db-d265b5711f33.mp4

#### After update/upgrade

https://user-images.githubusercontent.com/11048214/163727141-cab1ac79-ac87-4587-90f3-c494cb7a4ee3.mp4

#### Related Issue

https://github.com/pop-os/beta/issues/402

ronell1292k, 2022-04-17:
Just commenting this to confirm the same. Can reproduce this 1:1

cwsystem76, 2022-04-18:
Can confirm on an `oryp6` this morning. Will keep an eye on the PR.

PR #327 (2022-04-18): fix: Improve systemd service packaging
- Uses dh_installsystemd as debian recommends
- Makes service available as com.system76.PowerDaemon
- Fixes path of service file
- Improves packaging for #294

Fixes #332
closed 2022-04-18 by mmstick