pop-os/system76-power #334: Nvidia is not always pci device 01:00

lfom, 2022-04-25:
https://github.com/pop-os/system76-power/blob/bd282793f702ce3da12274691854eb87f829773b/src/daemon.rs#L232

I am not sure if this code is actually used somewhere else (I know very little Rust), but the Nvidia device is not always device 01:00 in the PCI bus. For instance, this is the result of an Acer Aspire 5 with GeForce MX350:
```
$ lspci | grep -i nvidia          
02:00.0 3D controller: NVIDIA Corporation GP107M [GeForce MX350] (rev a1)
```

crawfxrd, 2022-07-23:
It's only used for hotplug detection, which is only supported on System76 models.

Still, shouldn't hardcode the device path.

crawfxrd, 2022-12-20:
galp5 enumerates the device on bus 23:

```
23:00.0 3D controller: NVIDIA Corporation GA107M [GeForce RTX 3050 Mobile] (rev a1)
```