pop-os/system76-power #339: Hybrid mode should not use dGPU for the shell

lfom, 2022-05-04:
https://github.com/pop-os/system76-power/blob/98e7ad8f06ede4b8d49aee8da022369ad52d9ff0/src/graphics.rs#L22-L27

I think most users expect that in Hybrid mode dGPU is used only when required, so activating it to render the DE should not happen by default. Maybe this helps using the proprietary Nvidia drivers when using Wayland as well, since it will be used only for the apps that are started explicitly using the discrete graphics.

crawfxrd, 2022-05-04:
nvidia-drm is used for rendering. Xorg must hold a reference to the driver so it can be used (set by `/usr/share/X11/xorg.conf.d/10-nvidia.conf`). `modeset` just enables KMS.

If the shell is rendered on the dGPU, there is an entry for gnome-shell.

```
[oryp8 ~]$ system76-power graphics
nvidia
[oryp8 ~]$ nvidia-smi
Wed May  4 11:30:44 2022       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 470.86       Driver Version: 470.86       CUDA Version: 11.4     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0 Off |                  N/A |
| N/A   53C    P8    15W /  N/A |    193MiB / 16125MiB |      6%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      2309      G   /usr/lib/xorg/Xorg                 71MiB |
|    0   N/A  N/A      2484      G   /usr/bin/gnome-shell              119MiB |
+-----------------------------------------------------------------------------+
```

See that Xorg now uses 71 MiB of RAM instead of 4 MiB.

You can also check the runtime status of the GPU to see that it is put in a low power state when not used

```
$ cat /sys/bus/pci/devices/0000\:01\:00.0/power/runtime_status
suspended
```

lfom, 2022-05-04:
Thank you for the detailed explanation. So I guess that when KMS is enabled, then the discrete graphics is used on Wayland. I know it is not supported, but disabling it may prevent problems. Is there any reason to enable it in Hybrid mode?

crawfxrd, 2022-07-23:
Presumably only the NVIDIA graphics mode should require it. But then there's [this comment](https://forums.developer.nvidia.com/t/understanding-nvidia-drm-modeset-1-nvidia-linux-driver-modesetting/204068/3):

> All it really does is enable the `DRIVER_MODESET` capability flag in the nvidia-drm devices so that DRM clients can use the various modesetting APIs. In addition to allowing clients that talk to the low-level DRM interface to work, **it’s also necessary for some PRIME-related interoperability features.**

I guess we'd have to try and see.

crawfxrd, 2022-07-23:
I've opened #352 to test disabling modesetting in hybrid mode.

I'm closing this as resolved, since it's working as intended on Xorg and Wayland is not supported.

closed 2022-07-23 by crawfxrd