pop-os/system76-power #340: PC Freezes After Suspend

lucashtc, 2022-05-04:
**Distribution (run `cat /etc/os-release`):**
```
NAME="Pop!_OS"
VERSION="22.04 LTS"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 22.04 LTS"
VERSION_ID="22.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=jammy
UBUNTU_CODENAME=jammy
LOGO=distributor-logo-pop-os
```
 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
```
system76-power:
  Instalado: 1.1.20~1651689568~22.04~f010065
  Candidato: 1.1.20~1651689568~22.04~f010065
  Tabela de versão:
 *** 1.1.20~1651689568~22.04~f010065 1001
       1001 http://apt.pop-os.org/release jammy/main amd64 Packages
        100 /var/lib/dpkg/status
```

**Issue/Bug Description:**
Whenever I suspend my desktop and wake it back up, it becomes unresponsive. I always have to press the off button to turn off and on again my way out.
After disabling system76-power extension the problem seems to go away

**Steps to reproduce (if you know):**
Suspend the machine 

**Expected behavior:**
Suspend the machine and resume use after pressing the power button

**Other Notes:**

Solution found on Reddit: [Reddit](https://www.reddit.com/r/pop_os/comments/q5aiva/pc_freezes_after_suspend/)

phrogpilot73, 2022-05-05:
It appears to be an issue with the new kernel 5.17.5 and system76 power. When I have the extension enabled, it matches what you describe above. When I disable system76 power, it matches the description on the first post of [this reddit thread](https://www.reddit.com/r/pop_os/comments/uivt6i/kernel_517_update_breaks_suspend/). 
It is only repeatable on the 5.17.5 kernel, when rebooting into 5.16.19 - everything works as advertised.

lucashtc, 2022-05-05:
Today the problem reappeared.

lucashtc, 2022-05-05:
The problem reoccurred even with system76-power disabled.
So trying to see if the solution found at https://github.com/pop-os/shell/issues/1413#issuecomment-1118845050 will help.
So far I haven't had any problems.

phrogpilot73, 2022-05-06:
> The problem reoccurred even with system76-power disabled. So trying to see if the solution found at [pop-os/shell#1413 (comment)](https://github.com/pop-os/shell/issues/1413#issuecomment-1118845050) will help. So far I haven't had any problems.

I tried that, and it didn't work (maybe the system76-power extension works more low level than I thought). Mentioned it on a reddit thread, and a [system76 engineer](https://www.reddit.com/r/pop_os/comments/uivt6i/comment/i7fzykj/?utm_source=share&utm_medium=web2x&context=3) commented that it's an upstream bug related to bluetooth.

lucashtc, 2022-05-07:
But in my case the problem appeared with kernel 5.16.19
And every time this error occurred, there was no bluetooth device connected.

phrogpilot73, 2022-05-10:
> But in my case the problem appeared with kernel 5.16.19 And every time this error occurred, there was no bluetooth device connected.

My will suspend normally when I have bluetooth turned off, only. If no devices are connected - it still suspends then freezes.

lucashtc, 2022-05-23:
An observation:
A few minutes after locking the screen, it just crashes.
Not after I suspend.

lucashtc, 2022-05-23:
It's been a few days since I had this problem.
But I will continue to accompany

lucashtc, 2022-05-27:
I'm closing this issue, because I didn't have the problem anymore.

lucashtc, 2022-11-09:
Reopening as I thought I had it resolved, but the problem remains

lucashtc, 2022-11-09:
I can solve this problem by disabling this extension

Go into extensions and disable System76 Power extension

1. systemctl stop system76-power
2. systemctl disable system76-power
3. Reboot
Since the last updates, the problem seems to have gotten worse.

theodrake, 2023-03-08:
Interesting I have this issue on a System76 laptop. I have a reoccurring issue where Bluetooth just stops working. Only BT device I use is a mouse. I've updated the BT drivers and haven't seen that issue, but am seeing the issue with suspend.  Will try turning off BT then suspend.

theodrake, 2023-03-08:
Well just turned off BT and was able to suspend successfully. I was also able to turn back on BT and have it work. That was the other issue I was having. BT would turn off and even after turning it back on mouse wouldn't connect. Only solution I found was to reboot.