pop-os/system76-power #342: Sporadic lag spikes after updating to Pop!_OS 22.04

pushfoo, 2022-05-10:
**Distribution (run `cat /etc/os-release`):**
 ```
 NAME="Pop!_OS"
VERSION="22.04 LTS"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 22.04 LTS"
VERSION_ID="22.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=jammy
UBUNTU_CODENAME=jammy
LOGO=distributor-logo-pop-os
```
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
The affected user did not provide detailed information on this. I'm reporting it for them as I was considering purchasing a refurbished version of similar hardware.

As an overview:
1. Steam installed through apt
2. Native Linux games
3. Windows games under WINE/Proton

See next section for a quote with additional information.

**Issue/Bug Description:**
Updating to Pop!_OS 22.04 appears to have created short sporadic lag spikes when playing games:
> Hello, since i updated to 22.04 a few days ago i've been having a LOT of lag in games, as in periods of 3-4 seconds when i go from the usual fps to 5-10 fps. This happens regardless of how i'm running with games, so it happens with wine games (e.g. Maniaplanet) and native games (e.g. Portal 2). These 2 worked perfectly before the update and i admit that i am completely lost on what i should do now.

The following __temporary__ fix staves off the problem for an indeterminate amount of time through the power menu in the top bar:
1. Set the GPU mode to integrated
2. Reboot
3. Set the GPU mode to dedicated.
4. Reboot
5. Set the Power profile to high performance
6. Reboot if necessary

This is a chart of GPU usage from the affected user, with red underline indicating periods of dropped frame rate:
[image]

**Steps to reproduce (if you know):**
1. Use this hardware:
[image]
2. Update to 22.04
3. Launch games
4. Wait for lag spikes

**Expected behavior:**
No lag spikes, as with previous Pop!_OS versions.

**Other Notes:**

I was [directed to post this here by Brock on discord](https://discord.com/channels/845005478111346708/849721041797775370/971939234195832863) after attempting to help a user, @LinUwUxCat. I'm still not completely sure if it's a kernel issue or something to do with this repo.

This issue has also been discussed in the following locations:
1. [Discord](https://discord.com/channels/845005478111346708/971127602360225883)
2. [Reddit](https://www.reddit.com/r/pop_os/comments/uhley6/games_lagging_a_lot_after_updating_to_2204/)

LinUwUxCat, 2022-05-10:
Hello, I updated from 21.10, which was updated from 21.04 a few months before that.

LinUwUxCat, 2022-05-11:
I can also say that the lags stopped when i switched to Hybrid graphics yesterday, and everything was fine until right now, where i started getting lags again.
I don't think this is a kernel issue and i have no idea where this could come from.

pushfoo, 2022-05-16:
Wasn't upgrading to 510 drivers part of the fix, or did I misunderstand the discord discussion?

LinUwUxCat, 2022-05-16:
The problem was indeed fixed by upgrading the nvidia drivers from 470 to 510.

pushfoo, 2022-05-18:
I'm going to leave this open for now. My understanding is that each NVidia driver release drops support for some cards, which means that some users with older GPUs may be stuck on driver versions that recreate the problem.

LinUwUxCat, 2022-05-22:
Quick update, the lag spikes are still present while on NVIDIA-only mode, but not on hybrid mode (for now)

mmstick, 2022-05-22:
Moving this to the NVIDIA repository since it's NVIDIA driver related.