pop-os/system76-power #354: [regression] galp5 gpu management broken in 1.1.21+

ghthor, 2022-08-10:
**Distribution (run `cat /etc/os-release`):**

```shell
$ cat /etc/os-release 
NAME="Arch Linux"
PRETTY_NAME="Arch Linux"
ID=arch
BUILD_ID=rolling
ANSI_COLOR="38;2;23;147;209"
HOME_URL="https://archlinux.org/"
DOCUMENTATION_URL="https://wiki.archlinux.org/"
SUPPORT_URL="https://bbs.archlinux.org/"
BUG_REPORT_URL="https://bugs.archlinux.org/"
LOGO=archlinux-logo
Wed Aug 10 10:17:05 AM EDT 2022
```

```shell
$ sudo system76-driver-cli 
INFO	product_version: 'galp5'
INFO	remove_nvidia_dynamic_power_one: Remove NVIDIA dynamic power override
INFO	Skipping 'remove_nvidia_dynamic_power_one' as it was already applied
Wed Aug 10 10:29:07 AM EDT 2022
```
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**

## Working

```shell
$ yay -Qi system76-power
Name            : system76-power
Version         : 1.1.20-1
Description     : System76 Power Management
Architecture    : any
URL             : https://github.com/pop-os/system76-power
Licenses        : GPL
Groups          : None
Provides        : None
Depends On      : dbus  libusb  polkit
Optional Deps   : system76-acpi-dkms: some systems might need this driver
                  system76-dkms: some systems might need this driver
                  system76-io-dkms: some systems might need this driver
Required By     : None
Optional For    : system76-driver
Conflicts With  : None
Replaces        : None
Installed Size  : 2.42 MiB
Packager        : Will Owens <will@voltus.co>
Build Date      : Wed 10 Aug 2022 10:06:52 AM EDT
Install Date    : Wed 10 Aug 2022 10:08:21 AM EDT
Install Reason  : Explicitly installed
Install Script  : Yes
Validated By    : None
```

## Not Working

```shell
$ yay -Qi system76-power
Name            : system76-power
Version         : 1.1.21-1
Description     : System76 Power Management
Architecture    : any
URL             : https://github.com/pop-os/system76-power
Licenses        : GPL
Groups          : None
Provides        : None
Depends On      : dbus  libusb  polkit
Optional Deps   : system76-acpi-dkms: some systems might need this driver
                  system76-dkms: some systems might need this driver
                  system76-io-dkms: some systems might need this driver
Required By     : None
Optional For    : system76-driver
Conflicts With  : None
Replaces        : None
Installed Size  : 2.47 MiB
Packager        : Will Owens <will@voltus.co>
Build Date      : Wed 10 Aug 2022 10:00:06 AM EDT
Install Date    : Wed 10 Aug 2022 10:19:53 AM EDT
Install Reason  : Explicitly installed
Install Script  : Yes
Validated By    : None

Wed Aug 10 10:20:48 AM EDT 2022
```

**Issue/Bug Description:**

When I install 1.1.21 my system stops being able to manage the GPU state. When 1.1.20 is installed, my CPU temp goes down and I can set the GPU to integrated.

```shell
$ system76-power graphics power
off (discrete)
Wed Aug 10 10:21:57 AM EDT 2022
```

**Steps to reproduce (if you know):**

1. On a galp5 with the nvidia GPU install 1.1.21+
1. Set the GPU graphics to integrated
1. Reboot
1. Observe that the output of `system76-power graphics power` reports it is on
    ```shell
    $ system76-power graphics power
    on (discrete)
    ```
1. install 1.1.20
1. execute `system76-power graphics integrated` and reboot
1. Observe that the output of `system76-power graphics power` reports the dGPU is off
    ```shell
    $ system76-power graphics power
    off (discrete)
    ```

**Expected behavior:**

Installing 1.1.21+ should not result in the inability to manage the dGPU state.

**Other Notes:**

curiousercreative, 2022-08-24:
@ghthor what is the result of `system76-power graphics power off`? As long as I've been aware of the the graphics power command, it's always defaulted to on, but allows turning off.

KairanD, 2022-10-28:
I do have the same issue with an Asus laptop (K46CB, i7 3537U, GT 740M 128 bit). Even in integrated mode, 'system76-power graphics power' shows the dedicated card is on. The laptop gets very hot and the battery life is bad. Forcing the card to turn off (using 'system76-power graphics power off') shows the card as "off", but nothing seems to change in practice. Oddly enough, running in hybrid mode - which also keeps the card on - seems to generate less heat and improve battery life.

I was using bumblebee (bbswitch-dkms) and it was turning the card off completely, generating a lot less heat and almost doubling battery life. However, bbswitch-dkms doesn't compile for 5.18+ kernels (there's already a fix for Ubuntu 22.10, but Pop!_OS is going to keep the 22.04 LTS base for some time).

Using KDE Neon with bbswitch-dkms (kernel 5.15), my laptop idles around 42ºC. With Pop!_OS, it idles at 60ºC. Battery life in Neon is a lot better and the laptop doesn't feel hot to the touch.

This laptop is a secondary machine for me. I don't want to use the Nvidia card, I just need it turned off to be able to work. I love Pop!_OS, but system76-power is not able to turn off the card as bbswitch does.

ghthor, 2024-02-23:
I was able to use the suggestions from this comment to recover the battery life that I was losing from upgrading to 1.1.21+

1. https://github.com/pop-os/system76-power/issues/252#issuecomment-1094468060