pop-os/system76-power #358: Suspend broken on Gazelle 16 because of Nvidia GPU

JCDetwiler, 2022-08-27:
**Distribution (run `cat /etc/os-release`):** Arch Linux (5.19.4 / Nvidia driver 515.65.01)
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):** Related to using `xfce4-session-logout --suspend` with the Whisker Menu.

**Issue/Bug Description:** The system wakes up immediately after suspending. `journalctl -p 3 -b` reported logs that looked like [these](https://www.mail-archive.com/ubuntu-bugs@lists.ubuntu.com/msg5981417.html).

**Steps to reproduce (if you know):** You should able to run the command above to trigger it.

**Expected behavior:** The machine doesn't immediately wake up after screen tears.

**Other Notes:** I ended up finding my fix [here](https://www.mail-archive.com/ubuntu-bugs@lists.ubuntu.com/msg5981437.html).

# Known workaround:
I went into `/etc/modprobe.d/system76-power.conf` and modified this setting to be off instead of on.
```
options nvidia NVreg_PreserveVideoMemoryAllocations=0
```

## Proposed fix:
I think something in `src/graphics.rs` needs to be updated, possibly the S3 exception where this setting is being applied. That's the extent of the debugging I'm ready to put into this though, I hope it helps someone else.

agherzan, 2022-09-28:
Here are the relevant errors (oryx6 on my side):

```
onfiguring Power Management Support' section in the driver README.
[   86.105347] nvidia 0000:01:00.0: PM: pci_pm_suspend(): nv_pmops_suspend+0x0/0x30 [nvidia] returns -5
[   86.106085] nvidia 0000:01:00.0: PM: dpm_run_callback(): pci_pm_suspend+0x0/0x160 returns -5
[   86.106102] nvidia 0000:01:00.0: PM: failed to suspend async: error -5
[   86.690666] PM: Some devices failed to suspend, or early wake event detected
```

agherzan, 2022-09-28:
There is also a related known issue on the Nvidia side: https://download.nvidia.com/XFree86/Linux-x86_64/470.86/README/powermanagement.html#KnownIssuesAndWf438e

regulator-g, 2022-12-30:
Just want to say I have the same issue, I think quite a few people do, thanks for the workaround its not perfect but laptop is no longer frozen on resume