pop-os/system76-power #368: run time type conversion error

bennetyee, 2022-11-12:
https://github.com/pop-os/system76-power/blob/e0a6fb88c21e29393dc395fe3f15584c2852e3dc/src/args.rs#L107

$ system76-power charge-thresholds 60 80
thread 'main' panicked at 'Mismatch between definition and access of `thresholds`. Could not downcast to TypeId { t: 15007124987583238918 }, need to downcast to TypeId { t: 16733916091185954916 }
', /build/system76-power-BfpR1A/system76-power-1.1.24~1667870429~22.04~e0a6fb8~dev/vendor/clap/src/parser/error.rs:30:9
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

murtezayesil, 2022-11-20:
I am facing the same issue on `lemp11` with system76-power v1.1.24.

```
murteza@niagara:~$ RUST_BACKTRACE=full system76-power charge-thresholds 40 80
thread 'main' panicked at 'Mismatch between definition and access of `thresholds`. Could not downcast to TypeId { t: 4014619811965266330 }, need to downcast to TypeId { t: 16733916091185954916 }
', /build/system76-power-95IAuY/system76-power-1.1.24~1667870429~22.04~e0a6fb8/vendor/clap/src/parser/error.rs:30:9
stack backtrace:
   0:     0x55d8b393e7e0 - <unknown>
   1:     0x55d8b3963d1e - <unknown>
   2:     0x55d8b393b0a5 - <unknown>
   3:     0x55d8b3940013 - <unknown>
   4:     0x55d8b393fcff - <unknown>
   5:     0x55d8b39406ba - <unknown>
   6:     0x55d8b39405b7 - <unknown>
   7:     0x55d8b393ec8c - <unknown>
   8:     0x55d8b39402d2 - <unknown>
   9:     0x55d8b37d33d3 - <unknown>
  10:     0x55d8b381133c - <unknown>
  11:     0x55d8b381c629 - <unknown>
  12:     0x55d8b37d6498 - <unknown>
  13:     0x55d8b37d95a6 - <unknown>
  14:     0x55d8b37d63a3 - <unknown>
  15:     0x55d8b37d5479 - <unknown>
  16:     0x55d8b39365af - <unknown>
  17:     0x55d8b37d9838 - <unknown>
  18:     0x7efdca429d90 - <unknown>
  19:     0x7efdca429e40 - __libc_start_main
  20:     0x55d8b37d3665 - <unknown>
  21:                0x0 - <unknown>
```

bennetyee, 2022-11-20:
@murtezayesil 

it looks like QA is taking a while to get started looking at the fix.  maybe it's the holidays?

if you're desperate to set the charging thresholds meanwhile, you can either build from my branch, or just write directly into /sys/class/power_supply/BAT0/charge_control_{start,end}_threshold device files.  YMMV etc; my 2¢ -- i have no affiliation w/ system76 except as another customer / minor bugfix contributor.

murtezayesil, 2022-11-21:
Probably the holidays. I really hope they aren't following the footsteps of the tech giants and firing everyone :wink: 

`system76-power charge-thresholds --profile balanced` works just fine. I guess I will use that to set the limit at boot with systemd.

leviport, 2022-11-21:
We just have a lot going on. I'll take a look at this today. Thank you for your patience.

ahoneybun, 2022-12-09:
Should be fixed: https://github.com/pop-os/system76-power/pull/369

closed 2022-12-09 by ahoneybun