pop-os/system76-power #371: system76-power "panics" trying to set charge-thresholds

dubsygg, 2022-11-23:
**Distribution (run `cat /etc/os-release`):**
NAME="Pop!_OS"
VERSION="22.04 LTS"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 22.04 LTS"
VERSION_ID="22.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=jammy
UBUNTU_CODENAME=jammy
LOGO=distributor-logo-pop-os 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
system76-power:
  Installed: 1.1.24~1667870429~22.04~e0a6fb8
  Candidate: 1.1.24~1667870429~22.04~e0a6fb8
  Version table:
 *** 1.1.24~1667870429~22.04~e0a6fb8 1001
       1001 http://apt.pop-os.org/release jammy/main amd64 Packages
        100 /var/lib/dpkg/status

**Issue/Bug Description:**
Following latest updates, unable to set charge-thresholds via "system76-power", either through systemd service or by manual command-line entry. Error as follows:

`thread 'main' panicked at 'Mismatch between definition and access of `thresholds`. Could not downcast to TypeId { t: 4014619811965266330 }, need to downcast to TypeId { t: 16733916091185954916 }
', /build/system76-power-95IAuY/system76-power-1.1.24~1667870429~22.04~e0a6fb8/vendor/clap/src/parser/error.rs:30:9
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace`

**Steps to reproduce (if you know):**
1. Launch terminal;
2. issue command: "sudo system76-power charge-thresholds 85 95", Return;
3. 

**Expected behavior:**
It should set the charge thresholds on the battery and go about its day

**Other Notes:**
Posted issue here per recommendation from Brock in S76 Discord.

leviport, 2022-11-23:
>Looks to be the same reported issue as https://github.com/pop-os/system76-power/issues/368

So this is a duplicate report?

leviport, 2022-11-23:
No worries, I'll close this as a duplicate then. Since #369 has been merged, the fix should be released soon.

closed 2022-11-23 by leviport