pop-os/system76-power #380: Will not let me compile

PacoNoobxd, 2023-01-15:
sebas@Lenovo-15IMH05:~/Descargas/system76-power-1.1.24$ sudo make
[sudo] password for sebas: 
cargo build "--release"
error: failed to parse manifest at `/home/sebas/Descargas/system76-power-1.1.24/Cargo.toml`

Caused by:
  failed to parse the `edition` key

Caused by:
  this version of Cargo is older than the `2021` edition, and only supports `2015` and `2018` editions.

Installed from https://www.rust-lang.org/tools/install
and cargo with "sudo apt install cargo"

>>MY SYSTEM>>
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

crawfxrd, 2023-01-15:
Don't use `sudo` when compiling.

crawfxrd, 2023-01-20:
Closing this as fixed, since the issue is most likely using the distro provided cargo.

closed 2023-01-20 by crawfxrd