pop-os/system76-power #385: [Error] daemon returned error-  "does not have switchable graphics"

miikkuu, 2023-02-05:
daemon returned error-  "does not have switchable graphics" while trying to switch graphics with command :-

$ sudo system76-power graphics ...

OS:Ubuntu 22.04.1 LTS

I have Two Graphics Card in my System. 
[Amd and Intel Integrated]

But by default only the Integrated gfx is up.

$ lspci -nn | grep -E 'VGA|Display'
Returned Two Graphics as shown.

But using -: $ lspci  | grep VGA
Returned only one ! 
VGA : integrated

Why Can't I switch my gfx card?
How to solve this???

[[image]

[image]

[image]

Help Needed]

miikkuu, 2023-02-05:
CC @crawfxrd

jdileonardo, 2023-03-29:
experiencing the same, with a similar Intel iGPU and an AMD dGPU

julian@pop-os:~$ switcherooctl
Device: 0
  Name:        Intel® HD Graphics 530
  Default:     yes
  Environment: DRI_PRIME=pci-0000_00_02_0

Device: 1
  Name:        Advanced Micro Devices, Inc. [AMD®/ATI] Cape Verde PRO / Venus LE / Tropo PRO-L [Radeon HD 8830M / R7 250 / R7 M465X]
  Default:     no
  Environment: DRI_PRIME=pci-0000_01_00_0

jdileonardo, 2023-03-30:
Adding 
`
amdgpu.si_support=1 amdgpu.cik_support=1 radeon.si_support=0 radeon.cik_support=0`

to my kernel boot options allows me to use DRI_PRIME=1 before cmds now.

`julian@pop-os:~$ sudo gpu-manager
last_boot_file: /var/lib/ubuntu-drivers-common/last_gfx_boot
new_boot_file: /var/lib/ubuntu-drivers-common/last_gfx_boot
can't access /run/u-d-c-nvidia-was-loaded file
can't access /opt/amdgpu-pro/bin/amdgpu-pro-px
Looking for nvidia modules in /lib/modules/6.2.8-060208-generic/kernel
Looking for nvidia modules in /lib/modules/6.2.8-060208-generic/updates/dkms
Looking for amdgpu modules in /lib/modules/6.2.8-060208-generic/kernel
Looking for amdgpu modules in /lib/modules/6.2.8-060208-generic/updates/dkms
Is nvidia loaded? no
Was nvidia unloaded? no
Is nvidia blacklisted? no
Is intel loaded? yes
Is radeon loaded? yes
Is radeon blacklisted? no
Is amdgpu loaded? yes
Is amdgpu blacklisted? no
Is amdgpu versioned? no
Is amdgpu pro stack? no
Is nouveau loaded? no
Is nouveau blacklisted? yes
Is nvidia kernel module available? no
Is amdgpu kernel module available? no
Vendor/Device Id: 8086:191b
BusID "PCI:0@0:2:0"
Is boot vga? yes
Vendor/Device Id: 1002:682b
BusID "PCI:1@0:0:0"
can't open /sys/bus/pci/devices/0000:01:00.0/boot_vga
Is boot vga? no
can't open /sys/bus/pci/devices/0000:01:00.0/boot_vga
Found "/dev/dri/card1", driven by "amdgpu"
Number of connected outputs for /dev/dri/card1: 0
Skipping "/dev/dri/card1", driven by "amdgpu"
Skipping "/dev/dri/card0", driven by "i915"
Skipping "/dev/dri/card1", driven by "amdgpu"
Skipping "/dev/dri/card0", driven by "i915"
Skipping "/dev/dri/card1", driven by "amdgpu"
Found "/dev/dri/card0", driven by "i915"
output 0:
	card0-eDP-1
Number of connected outputs for /dev/dri/card0: 1
Does it require offloading? yes
last cards number = 2
Has amd? yes
Has intel? yes
Has nvidia? no
How many cards? 2
Has the system changed? No
Intel IGP detected
Desktop system detected
or laptop with open drivers
Nothing to do
`

```
julian@pop-os:~$ switcherooctl
Device: 0
  Name:        Intel® HD Graphics 530
  Default:     yes
  Environment: DRI_PRIME=pci-0000_00_02_0

Device: 1
  Name:        Advanced Micro Devices, Inc. [AMD®/ATI] Cape Verde PRO / Venus LE / Tropo PRO-L [Radeon HD 8830M / R7 250 / R7 M465X]
  Default:     no
  Environment: DRI_PRIME=pci-0000_01_00_0
```

miikkuu, 2023-03-31:
Any solution to this? It has been a while since I opened the issue.