pop-os/system76-power #388: oryp10: FanDaemon not starting, fan control seems to be not handled by system76-power daemon

r00t3g, 2023-02-21:
**Distribution (run `cat /etc/os-release`):**
NAME="Pop!_OS"
VERSION="22.04 LTS"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 22.04 LTS"
VERSION_ID="22.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=jammy
UBUNTU_CODENAME=jammy
LOGO=distributor-logo-pop-os
 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
system76-power:
  Installed: 1.1.24~1674505745~22.04~c504ff6
  Candidate: 1.1.24~1674505745~22.04~c504ff6
  Version table:
 *** 1.1.24~1674505745~22.04~c504ff6 1001
       1001 http://apt.pop-os.org/release jammy/main amd64 Packages
        100 /var/lib/dpkg/status

**Issue/Bug Description:**
Fan control does not work from system76-power. journalctl shows `system76-power[1542]: [ERROR] fan daemon: platform hwmon not found` after startup. As a result, FanDaemon is not running and does not do any fan control. Modifying fan curves in the code does not affect anything. Frankly speaking, I don't know what exactly controls the fans...

**Steps to reproduce (if you know):**
* Start or restart system76-power daemon using `systemctl restart com.system76.PowerDaemon.service`
* Check journal `journalctl -e -u com.system76.PowerDaemon.service` - observe `[ERROR] fan daemon: platform hwmon not found`
* Modify FanDaemon - put maximum fan speed to FanCurve::standard
* Add info-logging to FanDaemon::get_temp, set_duty methods
* Build a new binary, replace /usr/bin/system76-power with it
* Restart daemon
* Check journal - observe no added logging and no fan speed changes

**Expected behavior:**
* FanCurve change after modifying curves

**Other Notes:**
After a short discussion in https://github.com/system76/ec/issues/241#issuecomment-1351357188 regarding strange fan behavion on my oryp10, I have finally decided to tune it up and, surprisingly, I discovered that fan control with system76-power daemon doesn't seem to be working at all.

I have made a special fan curve for oryp10 and tried modifying all others - the was literally no effect with the new system76-power binary I have compiled and copied over the original one. 

Perhaps the problem is somehow related to /sys/class/hwmon, which has the following structure in my case:
```
lrwxrwxrwx  1 root root 0 Feb 21 11:29 hwmon0 -> ../../devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0003:00/power_supply/AC/hwmon0
lrwxrwxrwx  1 root root 0 Feb 21 11:29 hwmon1 -> ../../devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0/hwmon1
lrwxrwxrwx  1 root root 0 Feb 21 11:30 hwmon2 -> ../../devices/LNXSYSTM:00/LNXSYBUS:00/17761776:00/hwmon/hwmon2
lrwxrwxrwx  1 root root 0 Feb 21 11:30 hwmon3 -> ../../devices/pci0000:00/0000:00:06.0/0000:02:00.0/nvme/nvme0/hwmon3
lrwxrwxrwx  1 root root 0 Feb 21 11:31 hwmon4 -> ../../devices/platform/coretemp.0/hwmon/hwmon4
lrwxrwxrwx  1 root root 0 Feb 21 11:31 hwmon5 -> ../../devices/virtual/thermal/thermal_zone0/hwmon5
lrwxrwxrwx  1 root root 0 Feb 21 14:22 hwmon6 -> ../../devices/virtual/misc/uhid/0005:046D:B023.0003/power_supply/hidpp_battery_1/hwmon6
```

Especially the coretemp.0 link, which might be not properly interpreted into coretemp name by the sysfs-class Cargo package (sysfs-class = { git = "https://github.com/pop-os/sysfs-class" }).

Example of modyfied standard curve:
```rust
pub fn standard() -> Self {
    Self::default()
        .append(44_99, 100_00)
        .append(45_00, 100_00)
        .append(55_00, 100_00)
        .append(65_00, 100_00)
        .append(75_00, 100_00)
        .append(78_00, 100_00)
        .append(81_00, 100_00)
        .append(84_00, 100_00)
        .append(86_00, 100_00)
        .append(88_00, 100_00)
```

Any help would be appreciated as I get tired of listening for max-powered fans when the CPU is not even close to overheating. :)

jackpot51, 2023-02-21:
This code is for controlling the fan in System76 Thelio desktops. For laptop fan control on the oryp10, see https://github.com/system76/ec

closed 2023-02-21 by jackpot51

r00t3g, 2023-02-21:
@jackpot51 thanks. Sorry for misunderstanding. It appears that went the wrong way all the time :)