pop-os/system76-power #393: Unexpected PCIE Downgrades of the NVIDIA Gpu

RealAstolfo, 2023-04-30:
NAME="Pop!_OS"
VERSION="22.04 LTS"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 22.04 LTS"
VERSION_ID="22.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=jammy
UBUNTU_CODENAME=jammy
LOGO=distributor-logo-pop-os

 Issue occurred on Gentoo and PopOS, safe to assume it happens on any linux distro
 
 
Related Applications:
Any game utilizing the nvidia graphics card at a mid/high graphics cost

**Issue/Bug Description:**
Reporting as a bug, but im pretty sure its actually a feature of PCIE ASPM that is getting in the way.
The PCI Express lane downgrades from 4 x8 to 1 x8, causing sub par graphics performance and gpu% to maximimize

**Steps to reproduce (if you know):**
Play any game on the nvidia graphics card in fullscreen and wait for the PCI Express lane downgrade to occur. (EVEN ON AC)

**Expected behavior:**
No PCIE downgrade

**Other Notes:**
Fix/ Workaround

echo "on" > /sys/bus/pci/devices/0000\:01\:00.0/power/control
echo "on" > /sys/bus/pci/devices/0000\:01\:00.1/power/control
echo "0" > /sys/bus/pci/devices/0000\:01\:00.1/link/l0s_aspm
echo "0" > /sys/bus/pci/devices/0000\:01\:00.1/link/l1_1_aspm
echo "0" > /sys/bus/pci/devices/0000\:01\:00.1/link/l1_2_aspm
echo "0" > /sys/bus/pci/devices/0000\:01\:00.1/link/l1_aspm

The above has successfully fixed problems i mentioned above where games in fullscreen would suddenly be put down to sub-par FPS on the oryx pro (oryp10)

recommendation:
add the above commands to the system76-power performance profile

crawfxrd, 2023-05-15:
This should probably be in system76/firmware-open. ASPM can be disabled in coreboot.

RealAstolfo, 2023-05-16:
> This should probably be in system76/firmware-open. ASPM can be disabled in coreboot.

do you have additional details on how this option is implemented? i did find it kinda bizzare with the lack of options in the bios on my oryx pro considering what coreboot *should* have