pop-os/system76-power #394: Boot parameters are reset when switching graphics mode

dleviminzi, 2023-05-02:
**Distribution (run `cat /etc/os-release`):** 22.04 LTS
 
 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):** NA

**Issue/Bug Description:** Boot parameters are reset when changing graphics modes. I like to set `mem_sleep_default=deep` in `/boot/efi/loader/entries/Pop_OS-current.conf`. Unfortunately, if I switch graphics modes from integrated to discrete or do the reverse, the parameter will disappear. I then have to edit it again and reboot again, which is a bit of a pain. 

**Steps to reproduce (if you know):** Change some boot parameter and then switch graphics modes.

**Expected behavior:** I would think boot parameters would remain the same or at least revert back to what they were for each graphics mode. 

**Other Notes:** I'm not sure this is exactly the right place to report this. Sorry if it is not.

leviport, 2023-05-02:
Have you tried adding the boot parameter with Kernelstub instead of editing the boot entries directly?
```
sudo kernelstub -a mem_sleep_default=deep
```

Then I like to use `sudo kernelstub -p` to confirm that it was added correctly.

dleviminzi, 2023-05-03:
> Have you tried adding the boot parameter with Kernelstub instead of editing the boot entries directly?
> 
> ```
> sudo kernelstub -a mem_sleep_default=deep
> ```
> 
> Then I like to use `sudo kernelstub -p` to confirm that it was added correctly.

That solves it! Thanks for the tip, somehow didn't realize that existed.

closed 2023-05-03 by dleviminzi