pop-os/system76-power #398: Allow desktop computers to have switchable graphics

Mipsters, 2023-06-01:
**Distribution (run `cat /etc/os-release`):**

 ```
NAME="Pop!_OS"
VERSION="22.04 LTS"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 22.04 LTS"
VERSION_ID="22.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=jammy
UBUNTU_CODENAME=jammy
LOGO=distributor-logo-pop-os
```
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**

```
system76-power:
  Installed: 1.1.24~1684932166~22.04~4ebf3fd
  Candidate: 1.1.24~1684932166~22.04~4ebf3fd
  Version table:
 *** 1.1.24~1684932166~22.04~4ebf3fd 1001
       1001 http://apt.pop-os.org/release jammy/main amd64 Packages
        100 /var/lib/dpkg/status
```

**Issue/Bug Description:**

I have a desktop PC with a motherboard with onboard graphics and an nvidia GPU,
yet `system76-power` reports no switchable graphics

```console
$ system76-power graphics switchable
not switchable
```

**Steps to reproduce (if you know):**

have a PC with nvidia GPU and a motherboard with onboard graphics

install any `system76-power` version after 1.1.24~437d6ba

run the command

```sh
system76-power graphics switchable
```

get a report that you system is `not switchable`

**Expected behavior:**

A system with onboard and discrete GPUs will be able to switch between them, regardless if its a desktop or a laptop

**Other Notes:**

I am aware that disabling switchable graphics for desktop computers was a [deliberate design choice](https://github.com/pop-os/system76-power/commit/32c55a45a598eb4d3a3d5ec97ff78d44f7308298)  
but I got into a situation in which building a previous version of `system76-power` before disabling this feature solved my problem, so I think considering this a bug is appropriate

I have an nvidia 1070 in my PC, and I wished to use it and my motherboard's on board graphics card at the same time to have more screens connected to my PC

I have attempted many solutions, and got to very odd situations.

for example, when I used the system regularly, it defaulted to run using xserver
and the display I plugged to my motherboard was not detected at all

but forcing wayland on the system, made it able to detect the motherboard plugged display just fine

knowing that nvidia has issues with wayland (and experiencing some from games like Portal 2), I tried to see if there is a better solution  
online I only saw solutions involving `nvidia-prime`

after more research I discovered `system76-power` is the Pop!_OS equivalent, I attempted to use it for that purpose  
reading [Graphics Switching (Pop!_OS) ](https://support.system76.com/articles/graphics-switch-pop/) made it feel like Hybrid mode is just what I needed

and then I discovered that it was deliberately disabled...

and when I did manage to set hybrid mode with an old version of the tool, it seems like the system is set up in wayland only
so I'm sort of back to square one, but I didn't need to force it this time

this situation is very weird to me, and I don't think that what I'm trying to do is that outrageous
and it seems like allowing desktop PCs to have switchable graphics does result in a logical workflow

I would like to understand what is the reason it was disabled, and if there is a different solution to my problem

leviport, 2023-06-01:
>A system with onboard and discrete GPUs will be able to switch between them, regardless if its a desktop or a laptop

There's a fundamental difference between a desktop and a switchable graphics laptop, and that is that both the dGPU and iGPU have to share the built-in display, and sometimes even the display outputs. With a desktop, each GPU has its own outputs. If you want to switch GPUs on a desktop, you just have to plug into the other GPU's port (depending on your motherboard and BIOS, of course). Rendering offloading on a desktop GPU might not even be possible, as far as I'm aware. 

>but I got into a situation in which building a previous version of system76-power before disabling this feature solved my problem, so I think considering this a bug is appropriate

Could you elaborate a little more here? Were you actually able to use graphics switching on a desktop without changing which port the display was plugged into?

crawfxrd, 2023-06-01:
I can make it explicit in the README that switchable graphics is a feature designed for laptops and AIOs, but desktops are not supported.

> I have an nvidia 1070 in my PC, and I wished to use it and my motherboard's on board graphics card at the same time to have more screens connected to my PC

Doesn't the dGPU support like 4 outputs already?

Anyways, if your firmware supports it, using iGPU and dGPU outputs on desktop is normally controlled by a BIOS policy (something like "iGPU Multi-Monitor"). If your MB doesn't support that then you might not be able to use both as outputs at the same time.

Mipsters, 2023-06-03:
probably the difference I related to using an old version supporting switchable graphics originated from an error in my `/etc/X11/xorg.conf` file, sothis issue can be dismissed

but I do have an issue with Xserver and onboard graphic support
not changing any bios settings, switching to wayland manages to use the screen connected to the onboard graphics
but when using xserver, the display connected to onboard graphics does not register at all

Mipsters, 2023-06-03:
Solved this issue (partly, display behaves very oddly) by adding `Option "AllowExternalGpus"` in `/etc/X11/xorg.conf`