pop-os/system76-power #414: Add mux switching support for new models

MrPenguin07, 2023-09-17:
I see dp-alt mode isn't working on newer models, eg. darp8.
src/hotplug/mux.rs hasn't been updated to add support for recent models - is this due to a technical change in how the mux works?

I hesitate to add darp8 in my fork as I can't be sure the memory addresses or hardware control values are the same as previous models or whether this would work at all.

However dp-alt on the non TB usb-c port would be a welcome feature.

Advice?

jackpot51, 2023-09-17:
There is no displayport mux on any other models, which is why they are not using this code.

closed 2023-09-17 by jackpot51

MrPenguin07, 2023-09-18:
Thanks for that clarification.

Then where lies the issue in getting DP-alt over USB4? It worked with previous models using the mux here, the kernel should support it, recently adding DP 2.1 [https://lore.kernel.org/lkml/20230811210735.159529-1-utkarsh.h.patel@intel.com/](url) 

Or is this a hardware limitation on newer models?

jackpot51, 2023-09-18:
If that port is not actually connected to displayport lanes on the CPU, nothing can be done to get displayport from it. I see one HDMI port and one DisplayPort over USB-C port (that would be the ThunderBolt port) in the tech docs. The other USB-C port is USB only: https://tech-docs.system76.com/models/darp8/README.html