pop-os/system76-power #415: AMD P-State EPP doesn't change based on the selected power profile

knamroud, 2023-10-08:
**Distribution (run `cat /etc/os-release`):**
"Pop!_OS 22.04 LTS"
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
system76-power 1.1.25~1690395185~22.04~fc8eb49
kernel 6.5.4-76060504.202309191142~1695998943~22.04~070916d

**Issue/Bug Description:**
AMD P-State Active isn't working properly with system76-power.
Switching power profile won't reflect the change in the epp files, leading to a not so optimized experience.
I found this bug on a Ryzen 7 5700U and a 7 5825U.

**Steps to reproduce (if you know):**
Switch to amd_pstate active (epp).
Switch power profile.
[image]
Cat the various epp files and note how it is incoherent with the selected power profile:
[image]

**Expected behavior:**
Shouldn't the epp, as it is the main way in which we can control the performance of the CPU when using AMD P-State Active, change based on the power profile selected in system76-power? 
I also came across a merged pr that supposedly fixed this issue, but I'm still facing this issue on two different machines.

mmstick, 2023-10-08:
https://github.com/pop-os/system76-power/blob/master/src/cpufreq.rs#L22-L38

Check the value of `/sys/devices/system/cpu/cpu0/cpufreq/scaling_government` and `/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver`.

mmstick, 2023-10-09:
Try branch `epp` and see if that behaves how you expect.

PR #416 (2023-10-09): feat: amd-pstate-epp energy_performance_preference config support
Closes #415 

When the scaling governor is `amd-pstate-epp`, the energy-performance preference at `/sys/devices/system/cpu/cpu0/cpufreq/energy_performance_preference` should change when the power profile changes. The battery profile sets this to `balance_power`; the balanced profile to "balance_performance"; and the power profile to "performance".
closed 2023-10-09 by mmstick