pop-os/system76-power #417: SCSI/SATA link time power management breaks AHCI hotplugging on some motherboards

wdc-bob, 2023-10-12:
Hello! After seeing how nice power usage under pop-os is. I added system76 power daemon to a custom ubuntu distribution However, I noticed it is causing issues with AHCI drives not enumerating on some motherboards after hotplug. Namely an Asus P5Q-EM-D0 (yes I know it is ancient) 

**Distribution (run `cat /etc/os-release`):**
 root@wddt ➜ ~ cat /etc/os-release
PRETTY_NAME="Ubuntu 22.04.3 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.3 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy

**Issue/Bug Description:**
When com.system76.PowerDaemon is enabled, AHCI  hotplug no longer works. 
Even though "ahci.mobile_lpm_policy=1" is passed on the kernel commandline. 

**Steps to reproduce (if you know):**
1). Build  and install system76 power 
2). Enable system76 power 
3). reboot 
4). Hotplug and run "lsscsi" and see only the OS drive. All other drives need an rescan of the SCSI bus to show up. 

**Expected behavior:**
3).  reboot
4).  Hotplug and run "lsscsi" and see all drives that are attached to the system. 

**Other Notes:**
Seems other people have seemingly run into this as well without realizing it is likelty that this daemon is causing hotswap to not work for them: 
https://www.reddit.com/r/pop_os/comments/evl3kk/sata_hot_plug_not_working_on_popos_but_does_on/
https://www.reddit.com/r/pop_os/comments/gmptkg/usb3_and_hotswap_sata_set_as_swappable_not_seeing/

Really just need a way to configure the default behavior of enable/disable SATA/SCSI link power management. It is uneeded on these systems and tends to cause more issues than solves. 
I'd really like to use system76 power as it gives much better results than power-profiles-daemon. 
In the meantime, I am going to see if "libata.force=nolpm" kernel commandline helps and if not I plan to build system76 power with the SATA LPM commented out, if possible.

misfitbob, 2023-10-13:
"libata.force=nolpm" will make hotplug work if I scan the bus once, but unfortunately for my use case hot plug needs to work all the time.

misfitbob, 2023-10-13:
looking over the other issues, seems like a use case for #2