pop-os/system76-power #420: Cannot query or set charge thresholds on kernel 6.6

travissuel, 2023-12-04:
**Distribution (run `cat /etc/os-release`):** Fedora 39
 
 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):** system76-power 1.1.25

**Issue/Bug Description:**
When attempting to query or set the charge thresholds on kernel 6.6 (tried on 6.6.2 and 6.6.3), system76-power returns the error:
> daemon returned an error message: "Not running System76 firmware with charge threshold support"

Setting charge thresholds after booting kernel 6.5 works as expected, and the thresholds appear to persist when booting kernel 6.6 again. I've tried on both a darp8 and an oryp8 with the same results.

**Steps to reproduce (if you know):**
1. Install and boot kernel 6.6
1. Attempt to query and/or set charge-thresholds using system76-power

**Expected behavior:**
Successfully query or set charge-thresholds

crawfxrd, 2023-12-04:
Kernel 6.6 broke modules with a `color` attribute. system76-acpi is not loaded, so this won't work.

leviport, 2023-12-04:
The 6.6.2 version we're currently testing for release has that commit reverted: https://github.com/pop-os/linux/pull/290

System76-power seems fine with that kernel.

travissuel, 2023-12-14:
Issue resolved by kernel 6.6.6.

crawfxrd, 2023-12-14:
Fixed in 6.6.5 by upstream commit torvalds/linux@8f2244c9af24.

closed 2023-12-14 by crawfxrd