pop-os/system76-power #431: Can't change charge thresholds under kernel Linux 6.8.0-76060800daily20240311-generic

phrogpilot73, 2024-05-23:
**Distribution (run `cat /etc/os-release`):**
NAME="Pop!_OS"
VERSION="22.04 LTS"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 22.04 LTS"
VERSION_ID="22.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=jammy
UBUNTU_CODENAME=jammy
LOGO=distributor-logo-pop-os
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
system76-power:
  Installed: 1.2.0~1715198272~22.04~a546efa
  Candidate: 1.2.0~1715198272~22.04~a546efa
  Version table:
 *** 1.2.0~1715198272~22.04~a546efa 1001
       1001 http://apt.pop-os.org/release jammy/main amd64 Packages
        100 /var/lib/dpkg/status

**Issue/Bug Description:**
Kernel was updated to Linux 6.8.0-76060800daily20240311-generic, noticed charge thresholds were not working. The following is from the terminal:

cory@lemurpro:~$ system76-power charge-thresholds
org.freedesktop.DBus.Error.Failed: Not running System76 firmware with charge threshold support

Rebooted with Linux 6.6.10-76060610-generic kernel. The following is from terminal:

cory@lemurpro:~$ system76-power charge-thresholds
Profile: Balanced (balanced)
Start: 86
End: 90

**Steps to reproduce (if you know):**
Try to change charge-thresholds under kernel Linux 6.8.0-76060800daily20240311-generic, error persists. Can --list-profiles

**Expected behavior:**
Be able to change charge profiles, especially since I am on a lemp10, running 2023-20-13_5d997d1 firmware. Works fine on the old kernel.

crawfxrd, 2024-05-23:
Is `system76_acpi` loaded?

```
lsmod | grep system76
```

```
dkms status
```

phrogpilot73, 2024-05-23:
cory@lemurpro:~$ lsmod | grep system76
`system76_io            20480  0`

cory@lemurpro:~$ dkms status
`system76/1.0.14~1707324875~22.04~da04fd6, 6.6.10-76060610-generic, x86_64: installed
system76/1.0.14~1707324875~22.04~da04fd6, 6.8.0-76060800daily20240311-generic, x86_64: installed
system76_acpi/1.0.2~1707324865~22.04~19ad0c5, 6.6.10-76060610-generic, x86_64: installed (original_module exists)
system76_acpi/1.0.2~1707324865~22.04~19ad0c5, 6.8.0-76060800daily20240311-generic, x86_64: installed (original_module exists)
system76-io/1.0.3~1707324885~22.04~3dd4c32, 6.6.10-76060610-generic, x86_64: installed
system76-io/1.0.3~1707324885~22.04~3dd4c32, 6.8.0-76060800daily20240311-generic, x86_64: installed
virtualbox/7.0.10, 6.6.10-76060610-generic, x86_64: installed
virtualbox/7.0.10, 6.8.0-76060800daily20240311-generic, x86_64: installed
`

phrogpilot73, 2024-05-23:
Forced reinstallation of system76-acpi. It works now.

ahoneybun, 2024-05-23:
@phrogpilot73 which command did you use? With `dpkg`, `apt`, or something else?

jcs224, 2024-05-28:
I also ran into this. Was also able to fix just by downgrading my kernel to 6.6.10 with `kernelstub`. Should this be re-opened until this is fixed upstream?

crawfxrd, 2024-05-28:
Issue is not in system76-power.

You can fix it by rebuilding DKMS modules.

ahoneybun, 2024-05-28:
@crawfxrd that does not always fix the issue, can we move this to system76-acpi-dkms then? It seems to depend on which commands you run when it comes to DKMS if it will fix the issue or not.

crawfxrd, 2024-05-28:
Yes it does. Anything that uses `led_classdev` is broken because of this commit:

- https://github.com/pop-os/linux/commit/b57f055b1411061997700fb0c42bf340c3d045f4

This is a breaking change that was made without the kernel version changing, so DKMS rebuilds were not triggered.