pop-os/system76-power #433: `charge-thresholds` not working as expected

vasigorc, 2024-06-18:
**Distribution (run `cat /etc/os-release`):**

```shell
NAME="Pop!_OS"
VERSION="22.04 LTS"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 22.04 LTS"
VERSION_ID="22.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=jammy
UBUNTU_CODENAME=jammy
LOGO=distributor-logo-pop-os
```

**Issue/Bug Description:**

I've opened a System76 support ticket (#179593) but it was closed. I've the following power charge threshold:

```shell
➜  ✗ system76-power charge-thresholds
Profile: Custom
Start: 94
End: 100
```

At the time of me opening the support ticket my charging level was at 94% and showing "Not charging". It has now dropped to 93% and is in "pending charge":

```shell
➜  ✗ upower -d
Device: /org/freedesktop/UPower/devices/battery_BAT0
  native-path:          BAT0
  vendor:               Notebook
  model:                BAT
  serial:               0001
  power supply:         yes
  updated:              Tue 18 Jun 2024 02:23:32 PM EDT (115 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               pending-charge
    warning-level:       none
    energy:              101.445 Wh
    energy-empty:        0 Wh
    energy-full:         107.95 Wh
    energy-full-design:  103.056 Wh
    energy-rate:         0 W
    voltage:             16.821 V
    charge-cycles:       1
    percentage:          93%
    capacity:            100%
    technology:          lithium-ion
    icon-name:          'battery-full-charging-symbolic'

Device: /org/freedesktop/UPower/devices/line_power_AC
  native-path:          AC
  power supply:         yes
  updated:              Tue 18 Jun 2024 02:07:32 PM EDT (1075 seconds ago)
  has history:          no
  has statistics:       no
  line-power
    warning-level:       none
    online:              yes
    icon-name:          'ac-adapter-symbolic'

Device: /org/freedesktop/UPower/devices/DisplayDevice
  power supply:         yes
  updated:              Tue 18 Jun 2024 02:07:32 PM EDT (1075 seconds ago)
  has history:          no
  has statistics:       no
  battery
    present:             yes
    state:               pending-charge
    warning-level:       none
    energy:              101.445 Wh
    energy-full:         107.95 Wh
    energy-rate:         0 W
    charge-cycles:       N/A
    percentage:          93%
    icon-name:          'battery-full-charging-symbolic'

Daemon:
  daemon-version:  0.99.17
  on-battery:      no
  lid-is-closed:   no
  lid-is-present:  yes
  critical-action: HybridSleep
````

**Expected behavior:**

I would expect that battery charging would kick in at 94% and would stop charging at 100%. Support agent told me about battery trickling, but battery trickling should respect my selected profile. It does so on my other laptop (an Asus runniing Fedora 38 with KDE).

**Other Notes:**

I hope that this is correctly addressed. Please correct me otherwise.

jacobgkau, 2024-06-19:
> It does so on my other laptop (an Asus runniing Fedora 38 with KDE).

Charge thresholds are implemented in firmware. I see from your support ticket that you have a Bonobo WS 15, which runs [System76 Open Firmware](https://github.com/system76/firmware-open/). Your Asus machine would be running Asus's proprietary firmware. Differences in behavior likely stem from that.

To start with, I would like to explain that most laptops have either the battery or the EC firmware lie about the charge being 100% (they transparently reserve a bit for the health of the battery). Our batteries are "dumb" and our EC firmware does not lie about their actual charge, which is why we configure thresholds out-of-the-box. It's behaving similarly to other computers, but being more honest about it. If you really wanted to charge your Bonobo to 100% every time, you could disable charge thresholds altogether by running `system76-power charge-thresholds 0 100`, but we don't recommend this, because it's likely to kill your battery within a few years.

I do see that our support agents provided some incorrect information. The default start/stop values are actually 90%/100%. As proof, you can see this in the source code here: https://github.com/system76/ec/blob/2c5c7085694af5cd96fe4ce6465fcd98cfc3fb6d/src/board/system76/common/common.mk#L66 

The way thresholds work is that when you plug the system in, if it's below the start threshold, it will begin charging until it reaches the stop threshold. If it's not below the start threshold when plugged in, then it won't begin charging. That means if your start threshold is 94%, and you unplug the system and plug it back in at, say, 90%, then it should begin charging until it reaches the stop threshold (100%). If you plug the system in at, say, 97%, it won't begin charging since that's above the start threshold.

Now as for why you've currently set your start threshold to 94% but you're not charging at 93%, I'm not sure yet. Can you please provide the output of the following commands (you can copy/paste the whole block into a terminal to ensure we don't get any of them mixed up):

```
echo "Start threshold:"
cat /sys/class/power_supply/BAT0/charge_control_start_threshold
echo "End threshold:"
cat /sys/class/power_supply/BAT0/charge_control_end_threshold
echo "Percentage:"
cat /sys/class/power_supply/BAT0/capacity
echo "Current:"
cat /sys/class/power_supply/BAT0/current_now
echo "Status:"
cat /sys/class/power_supply/BAT0/status
```

This will help confirm at the firmware level what the status of the battery and charger is.

vasigorc, 2024-06-19:
Thanks for following-up @jacobgkau and providing detail!

> if your start threshold is 94%, and you unplug the system and plug it back in at, say, 90%, then it should begin charging until it reaches the stop threshold (100%). 

I've just tested it out and it didn't work. I saw the flash arrow on the battery logo for a few minutes (maybe a minute), but it disappeared leaving the percentage at 93% (what it was before).

Output for the suggested commands:

```shell
Start threshold:
94
End threshold:
100
Percentage:
93
Current:
0
Status:
Not charging
```

jacobgkau, 2024-06-21:
Thank you for the output. I just wanted to update you that I've been testing this in our quality assurance lab this week in conjunction with the support team. So far, I haven't been able to cause the battery percentage to actually go down at all while plugged in, both on a bonw15 with 4080 graphics (same as yours) or a gaze18 (which the support team said they'd previously seen discharge under load happen on). I've been trying to stress the systems in the hopes of overdrawing the charger and forcing it to start draining the battery, but (as should be the case) that is not happening.

I'm going to leave the bonw15 plugged in and running over the weekend to see if the battery percentage naturally drops at all. I drained it down to 96% before I started testing (then plugged it in) earlier this week, and it's still at 96% now. I currently have the battery thresholds set to the default 90/100, so the battery's not reaching the start threshold and the system is not charging the battery, it's just not draining (as the system's running off of the AC adapter).

I see from your upower output that your `charge-cycles` value is still `1`, so I believe you haven't discharged the battery below the start threshold at all the entire time you've had it (potential trickle-charging aside). However, I do want to confirm a few more things explicitly:
- You are seeing the battery percentage drain even when the system is plugged into power?
- Have you ever seen the percentage drop while you were sitting in front of the computer, or has it only happened while you were away from the system (and e.g. the power could have momentarily gone out and come back on without you noticing)?
- What is the timeframe you're seeing the percentage drop at? (E.g. it takes X hours to drop Y percent.)

I want to get answers for you about the behavior of the charging thresholds, but I have no way to test what happens when the percentage drops while plugged in if I can't get that to happen myself.

vasigorc, 2024-06-21:
Thanks @jacobgkau! You've outraced me by a few minutes with your comment 🙂  I'll close the ticket since

```shell
upower -d | grep %
    percentage:          100%
    capacity:            100%
    percentage:          100%
  ```
  
I was about to write that I though it was possible that there was an expected delay between unplugging and plugging back in the laptop before the charging kicks-in. I waited 5 seconds before and it didn't work, it also didn't start charging for longer than that: as it takes about a minute for me to get from basement to main floor at my place. However today I travelled to US, when I plugged the laptop in it was at 77% and it charged right up to 100%

But with what you've written about minimal threshold I think that it all lines-up. I'll ask System76 to close the support ticket and reset the thresholds to defaults (90 and 100).

closed 2024-06-21 by vasigorc

jacobgkau, 2024-06-21:
Glad to hear things seem to be working as expected! I am still interested in the threshold behavior.

It sounds like you actually did have the system unplugged for long enough to start draining percentages from the battery at certain points, and were not seeing it actually drain while plugged in? Did you ever drop it below the default 90% start threshold? Our firmware engineer told me this shouldn't be the case, but one explanation may be that new thresholds may not come into effect until a new charge cycle has started (which would require dropping below the old threshold).

I also do want to let you know for future reference that, as noted in the [support article](https://support.system76.com/articles/laptop-battery-thresholds), custom thresholds are currently reset when the EC is reset (which happens when the system is unplugged and powered off). If you keep the system powered on (including suspended) any time it's unplugged, then your thresholds would be preserved (and simply rebooting usually shouldn't reset the EC, even on battery power). However, if you actually power the system completely off while unplugged, that should already reset the thresholds back to 90/100.

vasigorc, 2024-06-22:
> Our firmware engineer told me this shouldn't be the case, but one explanation may be that new thresholds may not come into effect until a new charge cycle has started (which would require dropping below the old threshold).

Nope I don't think that the charge level ever went below 90% before. 93% seems to have been its lowest before today.

> However, if you actually power the system completely off while unplugged, that should already reset the thresholds back to 90/100.

Thank you for pointing that out: I did turn off the laptop and unplugged it, but not both together.