pop-os/system76-power #434: highly disappointed with the lemur pro fan noise

MehmetTopsakal, 2024-06-26:
I am a new owner of Lemur Pro with Core Ultra 7 155U CPU. 

Unfortunately, I am highly disappointed with the fan management of the laptop.

While idle, the fan is off (0rpm). If I do some work, fan starts to spin at a speed close to 9k rpm, then it stops. 

This is really annoying. I feel I am at an airport when the fan is running at 9k rpm. Why not fan runs at a constant low speed, say 1k, or 2k rpm? This was the case with all of my previous lenovo and mac laptops. 

https://support.system76.com/articles/fan-noise/ this article mentions "Fn+1 to toggle between the modes". But there is only two setting max fan speed (airport mode) and low speed (0 but quickly gets 9k later)

I think there should be more options for fan setting between 0 and 9k.

crawfxrd, 2024-06-26:
https://github.com/system76/firmware-open/issues/250

closed 2024-06-26 by crawfxrd

MehmetTopsakal, 2024-06-26:
My organization paid almost $3k to buy your product.

Instead of quickly marking this as "not planned", I would expect you to add
a few lines of explanatory info.

leviport, 2024-06-26:
It's not a matter of planning. We're always trying to improve our products. It's a matter of this issue being a duplicate of the one that was linked.

Our fan curves are tuned primarily with performance in mind. Since laptops (especially ultra portables like the Lemur Pro) have small heatsinks, the fan behavior reflects CPU load pretty closely. 

If you need it to be quieter and don't mind a bit of a performance reduction, you might want to try the "battery life" power profile. It doesn't make a massive difference currently, but it is noticeable. Further improvements to power profiles are in the works as well, so it'll only improve from its current state.

MehmetTopsakal, 2024-06-26:
Thanks for the info.

I saw the discussion on the linked post. Some say throwing out the computer out of window, some say rebuilding the firmware, some say "no possible way to configure it for stability with simple thresholds"...

I am aware of "batter life" power profile. It helps a little. Fun ran running constantly at ~2k rpm can be a solution. But it is not clear how to do it on the linked post.

rodrigo-ceccato, 2024-06-26:
@MehmetTopsakal A bit unrelated, but I have been happily using my Lemur with the ["undervolt" package ](https://github.com/georgewhewell/undervolt/tree/master)to control CPU usage. 

Somethign as simple as sudo `/home/itsme/Software/undervolt/.venv/bin/python -m undervolt --temp 79` allow me to use my laptop without making noise.
While it does not allow to edit the fan curves, it allow you to control your CPU power with a lot of granulraity.

MehmetTopsakal, 2024-06-26:
Thanks @rodrigo-ceccato I will try it

dsats7, 2025-11-07:
lmao the near-constant spooling up and down is pretty funny. Mine has started making a groaning sound for lack of a better word, and it kindof sounds like a slumbering giant breathing. Looked for a replacement fan and only see sus aftermarket ones on Amazon. In the documentation they say to contact support to start a parts purchase but man that seems nuts, I just want to click and buy instead of go back and forth for days over email.

@MehmetTopsakal  "I saw the discussion on the linked post. Some say throwing out the computer out of window, some say rebuilding the firmware, some say "no possible way to configure it for stability with simple thresholds"" haha indeed. Maybe they'll have some improvement to this in a couple years, but you can always throw it out the window today for free....

I don't quite know what to make of this laptop. In some ways it's made for power users, linux people, people who appreciate it being easy to debadge, and DIY/engineering types, but it also has terrible spare parts availability in the store, it's not particularly loaded up with I/O, no hardware power switch for camera/mic, the motherboard has a spot for a cellular modem and SIM tray but they aren't populated (and their supporting passives/ICs aren't either, so impractical to add one), and you can't do stuff like adjust the fan curves or otherwise get particularly deep into its software guts (afaik).

Back to the fan- surely they modeled the airflow, but the way the exhaust goes through that grille and then straight into the display hinge (and there is ~2.7mm of space between the outer surface of the grille and the hinge) doesn't feel right to me- if nothing else it's making more noise than needed because of all the disruptions to the flow, and dunno about the static pressure rating of the fan / the impact to air volume but it can't be nothing.

I kindof like this laptop, but yeah the fan hardware and software is a bit goofy.