pop-os/system76-power #445: [bug] add pop-up warning for battery being almost empty

feikedonia, 2025-01-18:
I was playing a video game and I had forgotten to re-plug-in the battery.
The laptop was at 2% when it suddenly went in to suspend. The problem is that at 2% the battery may die anyway, because my battery is a bit old. 

### Proposal
Add a warning massage triggering at a certain battery level.

cosmic-session-git 1.0.0.alpha.5.1.r0.g6e48e12-1

clarin-ebtio800090, 2025-01-18:
Hello, we tried to solve the issue.

This is what we did:

Implement a warning message that triggers at a certain battery level before the system suspends. This will give users a chance to save their work or plug in the charger.

You can review changes in this commit: https://github.com/clarin-ebtio800090/pop-os-system76-power-445/commit/480300230a87cd56f549f8d9abef549734e34ca3.

> [!CAUTION]
> Disclaimer: The concept of solution was created by AI and you should never copy paste this code before you check the correctness of generated code. Solution might not be complete, you should use this code as an inspiration only.

---

Latta AI seeks to solve problems in open source projects as part of its mission to support developers around the world. Learn more about our mission at https://latta.ai/ourmission . If you no longer want Latta AI to attempt solving issues on your repository, you can block this account.

feikedonia, 2025-01-18:
? 

What has my laptop battery to do with AI???