pop-os/system76-power #451: system76-power fails to recognize nct6799 fan controller via nct6775 on Fedora 42 (Thelio Mira R4)

lxmota, 2025-04-15:
I'm using a System76 Thelio Mira R4 running Fedora 42 with system76-power version 1.2.2-1, and I'm trying to restore fan control after upgrading from Fedora 41 to Fedora 42. But system76-power fails with the error:

`[ERROR] fan daemon: platform hwmon not found
`

This occurs even when the correct modules are loaded:

```
$ lsmod | grep -E 'nct6775|system76_acpi'
nct6775                49152  0
nct6775_core          114688  1 nct6775
system76_acpi          20480  0
```

My Super I/O chip is detected as nct6799, and it appears under /sys/class/hwmon/hwmon8/name.
I've also set the environment variable override like this:

```
# /etc/systemd/system/com.system76.PowerDaemon.service.d/override.conf
[Service]
Environment=SYSTEM76_POWER_FAN_HWMON=/sys/class/hwmon/hwmon8
```

But the error remains unchanged even after restart. I have verified that fan control works via firmware/BIOS (though aggressively loud) and that nct6799 is exposed through the nct6775 module and works with lm_sensors. But manual execution of
`sudo SYSTEM76_POWER_FAN_HWMON=/sys/class/hwmon/hwmon8 system76-power `
still logs the same `[ERROR] fan daemon: platform hwmon not found` message
System Details

1.   Machine: System76 Thelio Mira R4
2.   OS: Fedora 42 (fully updated)
3.   Kernel: 6.14.2-300.fc42.x86_64
4.   system76-power version: 1.2.2-1
5.   Fan controller: nct6799 via nct6775

Could support for nct6799 be added, or could the environment variable override be made to work correctly? Alternatively, is there a supported way to make system76-power use this hwmon device on non-Pop!_OS systems?

spcbfr, 2025-07-10:
Does this cause fans to spin at top speed at all times?
asking because I had that problem last time I tried to run something other than popos on my thelio and I want to know if the problem still occurs?

lxmota, 2025-07-10:
Yes, I’m traveling and I don’t remember exactly what I did. I’ll check when I return home.